emsesp/EMS-ESP #740: Flow temp increase for warm water missing

swa72, 2021-03-10:
I noticed that my RC300/GB192i provides a parameter to increase the set flow temp by xK (5...40) for warm water. However the EMS-ESP does not show this parameter? I could create a log of telegrams (how?), fiddle around with the parameter and post the logs.

German text of the parameter ... "Überhöhung der vom Wärmeerzeuger angeforderten Vorlauftemperatur zur Aufhei-
zung des Warmwasserspeichers. Die Grundeinstellung hängt vom installierten Wärmeerzeuger ab."

MichaelDvP, 2021-03-10:
If you change a parameter on the RC300 the new value is published to all or send to the device resposible for that value. 
Open telnet (with logging to file) and type `watch 10` to see all telegrams from/to the thermostat. Then change the parameter to different values on the thermostat.  Search the log for something like "Thermostat->all (id  xx), data yy, offset zz" for a thermostat setting or "Thermostat->Boiler..." for a boiler setting, where the data matches your changes.

swa72, 2021-03-10:
Ok, I could not set up putty to record anything. Anyway, I've changed the value from 20 to 19 to 21 to 20. Hope that helps.

```
ems-esp:/$ watch 10
Watching incoming telegrams, displayed in decoded format
Filtering only telegrams that match a device ID or telegram type of 0x10
003+10:18:39.558 N 48: [emsesp] Mixer(0x21) <- Thermostat(0x10), ?(0x292), data: 01 (offset 5)
003+10:18:39.585 N 49: [emsesp] Thermostat(0x10) -> Mixer(0x21), ?(0x292), data: 00 (offset 5)
003+10:18:43.183 N 50: [emsesp] Thermostat(0x10) -> Boiler(0x08), UBAParameterWW(0x33), data: 13 (offset 5)
003+10:18:48.559 N 51: [emsesp] Thermostat(0x10) -> Boiler(0x08), UBAParameterWW(0x33), data: 15 (offset 5)
003+10:18:52.736 N 52: [emsesp] Thermostat(0x10) -> Boiler(0x08), UBAParameterWW(0x33), data: 14 (offset 5)
003+10:19:03.011 N 53: [emsesp] Thermostat(0x10) -> Boiler(0x08), UBAFlags(0x35), data: 01 11
003+10:19:03.579 N 54: [emsesp] Thermostat(0x10) -> Me(0x0B), RC300Monitor(0x2A5), data: 00 DD 21 28 21 00 28 28 05 A0 03 03 03 05 A0 03 8A 00 00 11 01 03 08 A7 00
003+10:19:04.136 N 55: [emsesp] Thermostat(0x10) -> Me(0x0B), RC300Monitor(0x2A5), data: 06 04 00 00 00 00 FF 64 3C 00 3C 01 FF 01 02 (offset 25)
003+10:19:04.696 N 56: [emsesp] Thermostat(0x10) -> Me(0x0B), RC300Set(0x2B9), data: FF 2E 28 26 26 01 00 FF FF 08 2E 02 E1 20 01 0F 05 00 00 01 22
003+10:19:05.329 N 57: [emsesp] Thermostat(0x10) -> Me(0x0B), RC300Summer(0x2AF), data: 00 28 00 00 3C 2D 11 01 19 1E 00
```

MichaelDvP, 2021-03-11:
Ok, now we know what offset 5  in 0x33 is. I'll add it to v3. For mixed heatingcircuits there is a similar parameter, named `FlowTempOffset`, i will add this as `wWFlowTempOffset` for reading and writing.

swa72, 2021-03-11:
Thanks, here is another one. My boiler has different settings to limit the power.

`sensor.boiler_burner_selected_max_power` (in HA speak) set by the system depending on use case
`sensor.boiler_burner_max_power` (in HA speak) set by user (not relevant for warm water); aka "MAX.HEIZLEISTUNG"

??? set by user to limit power for use case warm water; aka "MAX.WW-LEISTUNG"
I could not see/set this (maybe due to my system)

??? set by user to limit power overall; aka "Obergrenze Maximal Heizleistung"
I could not see/set this and changed the setting from 89 to 88 to 90.
```
ems-esp:/# watch 10
Watching incoming telegrams, displayed in decoded format
Filtering only telegrams that match a device ID or telegram type of 0x10
004+01:37:48.942 N 1: [emsesp] Thermostat(0x10) -> Boiler(0x08), ?(0x26), data: 58 (offset 7)
004+01:37:53.823 N 2: [emsesp] Thermostat(0x10) -> Boiler(0x08), ?(0x26), data: 5A (offset 7)
```

MichaelDvP, 2021-03-11:
Telegram 0x26 is first seen now. Can you do a `read 9 26` to fetch the complete telegram. 
(This telegram is unknown to my boiler.)

proddy, 2021-03-11:
read 8 ...

swa72, 2021-03-12:
```
ems-esp:/# read 8 26
000+00:02:27.284 N 2: [emsesp] Boiler(0x08) -> Me(0x0B), ?(0x26), data: 01 05 00 0F 00 1E 58 5A
```

MichaelDvP, 2021-03-12:
@swa72 Thanks for the telegram. I've added the offset 7 as wwMaxPower (i have no idea for the other values in that telegram). Please try the esp32_dev and check the values and set commands for wwFlowTempOffset and wwMaxPower.

proddy, 2021-04-17:
@swa72 can we close this?

closed 2021-10-04 by proddy