emsesp/EMS-ESP #744: Make the port for syslog server selectable

tp1de, 2021-03-11:
I would like to enter a different port then udp port 514 for the syslog server. 
Next to the ip-address of the server the respective port should be possible to select.

E.g.: 192.168.178.100 same as 192.168.178.100:514  or a different port: 192.168.178.100:12345

tp1de, 2021-03-12:
@proddy Is there any temporary solution that I can select a specific port for syslog - e.g. 20514 for node-red syslog-node?

proddy, 2021-03-12:
port forwarding? :-)

but seriously, it's not too hard to change. I'll add the feature tonight.

tp1de, 2021-03-12:
> but seriously, it's not too hard to change. I'll add the feature tonight.

This sounds good - Thanks for your support. 

I want to use the syslog-node in node-red and node-red does not run under root permissions. 
So I need to setup portnumbers above 1024 ......

proddy, 2021-03-12:
added it, but not tested. can you see if it works?

tp1de, 2021-03-12:
> can you see if it works?

@proddy   Yes it is working fine. Thanks a lot

closed 2021-03-12 by proddy