emsesp/EMS-ESP #745: setting thermostat time

dia-de-muertos, 2021-03-11:
Hello,
I'm trying to set the thermostat time via the web API (preferable NTP),
but neither NTP nor setting the time works:

```
$ date && curl "http://192.168.192.50/api?device=thermostat&cmd=datetime&data=NTP" && echo "" && curl -s "http://192.168.192.50/api?device=thermostat&cmd=info" | grep time
Thu 11 Mar 2021 03:21:26 PM CET
OK
  "time": "15:20:29 11/03/2021",

```

```
$ date && curl "http://192.168.192.50/api?device=thermostat&cmd=datetime&data=15:22:00-11.03.2021-3-0" && echo "" && curl -s "http://192.168.192.50/api?device=thermostat&cmd=info" | grep time
Thu 11 Mar 2021 03:21:39 PM CET
OK
  "time": "15:20:36 11/03/2021",
```

Did I understand the parameter wrong or is it a bug?

Running Version 2.2.0 on esp8266:
{
  "System": {
    "version": "2.2.0",
    "uptime": "016+00:04:14.947",
    "freemem": 48,
    "fragmem": 17
  },
  "Settings": {
    "enabled": "off",
    "publish_time_boiler": 10,
    "publish_time_thermostat": 10,
    "publish_time_solar": 10,
    "publish_time_mixer": 10,
    "publish_time_other": 10,
    "publish_time_sensor": 10,
    "mqtt_format": 2,
    "mqtt_qos": 0,
    "mqtt_retain": "off",
    "tx_mode": 1,
    "ems_bus_id": 11,
    "master_thermostat": 0,
    "rx_gpio": 13,
    "tx_gpio": 15,
    "dallas_gpio": 14,
    "dallas_parasite": "off",
    "led_gpio": 2,
    "hide_led": "off",
    "api_enabled": "on",
    "bool_format": 1,
    "analog_enabled": "off"
  },
  "Status": {
    "bus": "connected",
    "bus protocol": "Buderus",
    "#telegrams received": 903723,
    "#read requests sent": 299650,
    "#write requests sent": 41,
    "#incomplete telegrams": 1,
    "#tx fails": 3,
    "rx line quality": 100,
    "tx line quality": 100,
    "#MQTT publish fails": 0,
    "#dallas sensors": 3
  },
  "Devices": [
    {
      "type": "Boiler",
      "name": "BK13/BK15/Smartline/GB1x2 (DeviceID:0x08, ProductID:64, Version:02.07)",
      "handlers": "0x10 0x11 0x14 0x15 0x16 0x18 0x19 0x1A 0x1C 0x2A 0x33 0x34 0x35 0xD1 0xE3 0xE4 0xE5 0xE6 0xE9 0xEA 0x494 0x495"
    },
    {
      "type": "Thermostat",
      "name": "RC30 (DeviceID:0x10, ProductID:67, Version:02.08)",
      "handlers": "0xA3 0x06 0xA2 0x12 0x3E 0x3D 0x3F 0x48 0x47 0x49 0x52 0x51 0x53 0x5C 0x5B 0x5D 0xA5 0x37"
    },
    {
      "type": "Controller",
      "name": "BC10/RFM20 (DeviceID:0x09, ProductID:68, Version:02.00)",
      "handlers": ""
    }
  ]

MichaelDvP, 2021-03-12:
For me it works, tested on RC35 and v3, but i think there is no change since v2.2.0. 
Can you check with telnet, `log all` and `call thermostat datetime NTP` and post the output here. Maybe RC30 needs other settings for the last flag byte.

dia-de-muertos, 2021-03-12:
Hm, everything seems ok:
```
ems-esp:/# call thermostat datetime NTP
000+00:09:14.814 I 419: [thermostat] Date and time: 2021-03-12T14:58:32+0100
000+00:09:14.814 I 420: [thermostat] Setting date and time
000+00:09:14.897 D 421: [telegram] Sending write Tx [#150], telegram: 0B 10 06 00 15 03 0E 0C 3A 20 04 02 0D
000+00:09:15.045 D 422: [emsesp] Last Tx write successful
000+00:09:15.045 D 423: [telegram] Sending post validate read, type ID 0x06 to dest 0x10
000+00:09:15.295 D 424: [telegram] Sending read Tx [#151], telegram: 0B 90 06 00 20 6C
000+00:09:15.446 D 425: [emsesp] Last Tx read successful
000+00:09:15.446 T 426: [emsesp] Thermostat(0x10) -> Me(0x0B), RCTime(0x06), data: 15 03 0E 0C 39 24 04 00
```

But when I do a  `show values`,  the time is still behind:
```
Thermostat: RC30 (DeviceID:0x10, ProductID:67, Version:02.08)
  Time: 14:57:36 12/03/2021
```

MichaelDvP, 2021-03-12:
Yes, the readback gives the old time. You can try a bit with the last byte, it's a bitfield bit0 summertime, bit1 external clock setting. For my RC35 (and other RC3x0) the bit have to be set, otherwise the clock is not set. Maybe for RC30 the bit should be 0.
From NTP you get a time message:
`000+00:09:14.897 D 421: [telegram] Sending write Tx [#150], telegram: 0B 10 06 00 15 03 0E 0C 3A 20 04 02 0D`
then try sending manual without the last crc and setting the 02 to 00:
`call system send "0B 10 06 00 15 03 0E 0C 3A 20 04 00"`
and see if it works.

dia-de-muertos, 2021-03-12:
```
ems-esp:/#  call system send "0B 10 06 00 15 03 0E 0C 3A 20 04 00"
000+01:33:54.129 D 4: [telegram] Sending write Tx [#220], telegram: 0B 10 06 00 15 03 0E 0C 3A 20 04 00 0F
000+01:33:54.213 D 5: [emsesp] Last Tx write successful
000+01:33:54.213 D 6: [telegram] Sending post validate read, type ID 0x06 to dest 0x10
000+01:33:54.377 D 7: [telegram] Sending read Tx [#221], telegram: 0B 90 06 00 20 6C
000+01:33:54.491 D 8: [emsesp] Last Tx read successful
000+01:33:54.491 N 9: [emsesp] Thermostat(0x10) -> Me(0x0B), RCTime(0x06), data: 15 03 10 0C 16 0F 04 00
```

No success, time is still not set :-(

MichaelDvP, 2021-03-12:
Sad, seems the clock can not be set in RC30. Last chance:
`watch on` and change the time and date on the thermostat. Maybe there is something in the telegrams, but i'm not confident.

dia-de-muertos, 2021-03-12:
don't know if I did it right, but here probably what happend:
```
000+02:29:40.755 N 3: [emsesp] Thermostat(0x10) -> Boiler(0x08), UBASetPoints(0x1A), data: 23 64 64 00
000+02:29:54.283 N 10: [emsesp] Thermostat(0x10) -> All(0x00), RC35Monitor(0x3E), data: 04 02 28 00 E5 1A 3C 23 32 3F 00 00 64 11 23
000+02:29:54.505 N 11: [emsesp] Thermostat(0x10) -> Boiler(0x08), UBAFlags(0x35), data: 11 11
000+02:29:55.145 N 13: [emsesp] Thermostat(0x10) -> All(0x00), RCTime(0x06), data: 15 03 11 0C 12 16 04 00
000+02:30:01.408 N 18: [emsesp] Thermostat(0x10) -> Me(0x0B), RC35Monitor(0x3E), data: 04 02 28 00 E5 1A 3C 23 32 3F 00 00 64 11 23
000+02:30:01.818 N 19: [emsesp] Thermostat(0x10) -> Me(0x0B), RC35Set(0x3D), data: 01 25 28 22 02 28 00 02 00 05 05 2D 01 01 04 4B 05 4B 01 FF 3C FF 0F FA 05 02 02
000+02:30:02.162 N 20: [emsesp] Thermostat(0x10) -> Me(0x0B), RC35Set(0x3D), data: 02 02 00 01 0F 32 0A (offset 25)
000+02:30:02.444 N 21: [emsesp] Thermostat(0x10) -> Me(0x0B), RC35Timer(0x3F), data: 01 24 00 7E 21 24 20 7E 41 24 40 7E 61 24 60 7E 81 24 80 81 A1 30 A0 81 C1 30 C0
000+02:30:02.948 N 22: [emsesp] Thermostat(0x10) -> Me(0x0B), RC35Timer(0x3F), data: 30 C0 7E E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 (offset 25)
000+02:30:03.323 N 23: [emsesp] Thermostat(0x10) -> Me(0x0B), RC35Timer(0x3F), data: E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 (offset 50)
000+02:30:04.062 N 26: [emsesp] Thermostat(0x10) -> Me(0x0B), RC35Timer(0x3F), data: 90 E7 90 E7 90 E7 90 E7 90 00 00 00 01 01 00 01 01 00 01 01 00 01 01 00 (offset 75)
000+02:30:04.408 N 27: [emsesp] Thermostat(0x10) -> Me(0x0B), IBASettings(0xA5), data: 00 00 00 00 FF F1 00 06 00 01 0D 01 00
000+02:30:04.649 N 28: [emsesp] Thermostat(0x10) -> Me(0x0B), WWSettings(0x37), data: 00 00 02 02 00 01 01 00 3C
000+02:30:09.135 N 29: [emsesp] Thermostat(0x10) -> All(0x00), RCOutdoorTemp(0xA3), data: 0A 01 01
```

MichaelDvP, 2021-03-12:
It's only that the thermostat publishes the new time as
`Thermostat(0x10) -> All(0x00), RCTime(0x06), data: 15 03 11 0C 12 16 04 00`
meaning 12.03.2021, 17:18:22, 4.day of week, DST off. 
Sorry, i think the clock can not be set via bus.

dia-de-muertos, 2021-03-12:
Sad to hear :-(
Thanks for your fast and professional help!

closed 2021-03-12 by dia-de-muertos