emsesp/EMS-ESP #746: temporaryRoomSetpoint per hc similar to km200 ist not available as datafield and does not work properly

tp1de, 2021-03-11:
In km200 there are two seperate datafields for each hc available:

currentRoomSetpoint -> ems-esp: seltemp (read only)
temporaryRoomSetpoint -> ems-esp: n.a. - writeable with the temp command but works only partially. 

In km200 the temporaryRoomSetpoint can have the following values:
```
-1    : Automatic mode
0     : hc is switched off
5-30  : temperature setpoint in °C for hc
```

The setpoint is valid until next switchpoint from active switchprogram when in automatic mode.
This is very convenient to use, since it sets back automatic to time program mode. (my most used value to change)

The ems-esp command temp works in a way that 0 and 5-30°C can be set.
The setting back to Automatic mode doesn't work. The value -1 is represented as 5°C wich is wrong. 

It is important to introduce the new datapoint for a temporaryRoomSetpoint for the interfaces (mqtt, ebui, web api etc) and to make it properly work.

I did some debugging by changing values by km200 and watching on console:

```
Gateway(0x48) -> Thermostat(0x10)
hc1: RC300Set(0x2B9) offset 8
hc2: RC300Set(0x2BA) offset 8
value 0 : data: 00
value -1: data: FF
value 20: data: 28
```

@MichaelDvP @proddy 
You have commented in #737 and you are aware. Do we share the same understanding that adjustment ist needed?

proddy, 2021-03-11:
I haven't really been paying a lot of attention to this issue (was busy with HA MQTT stuff). I don't think its a bug, but happy to include it. I think Michael knows more about this one.

MichaelDvP, 2021-03-12:
I really don't understand the problem. What i understand:
KM200 uses a single field to set temperture and mode. And you have to enter some strange and not logical numbers to change the mode. 
Ems-esp have a mode  command, why not use this? 

But mode is set in offset 0, auto is indeed value 0xFF. If someone wants to set 0xFF to offset 8, which is the temperature field set the temp to 127.5, since the field is doubled. `-1` as signed int in a doubled field would set 0xFE!
This is confusing to me.

tp1de, 2021-03-12:
> I really don't understand the problem. What i understand:
> KM200 uses a single field to set temperture and mode. And you have to enter some strange and not logical numbers to change the mode.
> Ems-esp have a mode command, why not use this?

@MichaelDvP Let me try to explain again:

In km200 there are the same datafields like in ems-esp for mode, manualtemp, comforttemp etc. - just different names.
The field temporaryRoomSetpoint is an **additional** field. By changing values **the mode is not changed**.
**The fieldvalue is reset to automatic mode (-1) when next switchpoint from time program is reached**. 
Therefore it is a field with a **temporary value**. The value can be set by own webui or by programm and keeps the value until being reset automatically by time-program when in automatic mode. So the datafield can always be displayed and changed in own webui.

At the end it is the same behavior like on my radiator thermostats from homematic - controlled by automatic mode. 
When temperature at radiator is manually changed, this change s valid until next timeprogram switchpoint. 
Same logic for km200 temporaryRoomSetpoint.
Big advantage from my point of view: **Any changes done are automatically reset** - even if one forgets to reset manually or the program is not running / working anymore. 

Going to manual mode is certainly possible (as well on km200) but you miss the advantages of automatic reset. I recognized that my family members often forget to reset the temperature, so we do not use manual mode anymore on the homematic thermostats as well as on the central heating system.

By the way the **command temp for device thermostat** is changing the `temporaryRoomSetpoint` on km200. 
Just resetting manually by vommand to automatic mode (-1 ) is not working. (0xFF)
**And an own datapoint in mqtt or webui is missing**. 
It should be available in ems-esp as well, so that it can be displayed and changed. I would recommend not to use `temp` but rather like existing seltemp, currtemp, ecotemp etc. keeping this logic a fieldname like `temporarytemp` 

I hope I could clarify a bit.

_I personally believe that at the end ems-esp should cover km200 functionality and adds additional functions like the control parameters, mqtt integration etc._

MichaelDvP, 2021-03-12:
> (-1 ) is not working. (0xFF)

I've wrote it before: -1 is not 0xFF in a settemperature field with factor 2! To set 0xFF you have to enter `call thermostat temp 127.5` Why you do not try it and, if not working, send a debug log.

> I would recommend not to use temp but rather like existing seltemp, currtemp, ecotemp etc. keeping this logic a fieldname like temporarytemp

In auto-mode the temp command always sets temporary temperature, i don't see any benefit in doubling fileds.

I'll add a logic that the temp command for RC300 in auto mode will send 0xFF to ems if temp set to -1.

tp1de, 2021-03-12:
> In auto-mode the temp command always sets temporary temperature, i don't see any benefit in doubling fileds.

So far so good and 127.5 is working **but there is no datafield in mqtt or webui dashboard for temp** to indicate the value set by the command.

A  -1 value stays as a -1 value in km200 to indicate the last command sent - even if seltemp is set to 21. 
E.g. if a 20 is send it stays at 20 to indicate that the value is temporarily overwritten and set back to -1 on next time switchpoint.
Anyhow in most cases in ems-esp the datafield and command name is identical. This is good and similar to km200 logic.
It should be the same in this case.

I just thougth that `temp` as fieldname  is a bit misleading.....(temporary vs. temperature)

tp1de, 2021-03-12:
For info: Each of the fields returned by km200 rest-api have the following definitions. E.g.:

  "native": { "km200": {"id": "/heatingCircuits/hc1/temporaryRoomSetpoint",
      "type": "floatValue",
      "writeable": 1,
      "recordable": 0,
      "value": 5,
      "unitOfMeasure": "C",
      "minValue": 5,
      "maxValue": 30,
      "state": [ {"off": 0}, {"na": -1 } ]
      "valIs": "value" } }

Don't you think that for identical fields km200 and ems-esp should have same content?

glitter-ball, 2021-03-12:
Read this with interest. I regularly use the climate entity in HA to temporarily change the setpoint temp on my thermostat. I dial the control to the temp I want and it changes the setpoint until the next timed programme change, when it reverts to the programme. I don't touch the auto/manual/off icons, just change the temperature.

Does that help this at all?