emsesp/EMS-ESP #750: flowSetTemp missing decimal precision in values for FW200

TomoKRK, 2021-03-19:
**Bug description**
flowTempSet readouts for mixer devices are only showing integer values and are missing the decimal point.
I have been trying to nail it down and based on the values observed on my FW200 and monitoring the transmission
between FW200 and mixer circuits, I think I have isolated this to some telegrams which are carrying this value.

Here are the examples of those telegrams:
 
1) That's for my floor heating circuit where the setFlowTemp readout on FW200 says 26.1 ℃
000+02:45:59.930 N 657: [emsesp] Thermostat(0x10) -> Mixer(0x21), ?(0x23), data: 1A 64 00
90 21 23 00 1A 64 00 89

2) That's telegram for my radiator circuit where the setFlowTemp on FW200 says 38.1 ℃
90 20 23 00 26 64 00 59

**Expected behavior**
Get flowSetTemp readouts with decimal point i.e. 26.1

**Device information**
{
  "System": {
    "version": "2.2.1",
    "uptime": "000+03:12:13.846",
    "freemem": 42,
    "fragmem": 26
  },
  "Status": {
    "bus": "connected",
    "bus protocol": "HT3",
    "#telegrams received": 11604,
    "#read requests sent": 1703,
    "#write requests sent": 0,
    "#incomplete telegrams": 254,
    "#tx fails": 43,
    "rx line quality": 100,
    "tx line quality": 97,
    "#MQTT publish fails": 0,
    "#dallas sensors": 0,
    "#dallas fails": 0
  },
  "Devices": [
    {
      "type": "Boiler",
      "name": "Condens 2500/Logamax/Logomatic/Cerapur Top/Greenstar/Generic HT3 (DeviceID:0x08, ProductID:95, Version:18.08)",
      "handlers": "0x10 0x11 0x14 0x15 0x16 0x18 0x19 0x1A 0x1C 0x2A 0x33 0x34 0x35 0xD1 0xE3 0xE4 0xE5 0xE6 0xE9 0xEA 0x494 0x495"
    },
    {
      "type": "Thermostat",
      "name": "Junkers FW200 (DeviceID:0x10 ProductID:106, Version:12.12)",
      "handlers": "0xA3 0x06 0xA2 0x12 0x16F 0x170 0x171 0x172 0x165 0x166 0x167 0x168"
    },
    {
      "type": "Mixer",
      "name": "Junkers IPM (DeviceID:0x20 ProductID:102, Version:20.07)",
      "handlers": "0x10C"
    },
    {
      "type": "Mixer",
      "name": "Junkers IPM (DeviceID:0x21 ProductID:102, Version:20.07)",
      "handlers": "0x10C"
    },
    {
      "type": "Controller",
      "name": "HT3 (DeviceID:0x09, ProductID:95, Version:18.08)",
      "handlers": ""
    }
  ]
}

proddy, 2021-03-19:
the `setFlowTemp` comes from the Boiler right? Is the value incorrect there when you do a http://ems-esp/api?device=boiler&cmd=info

MichaelDvP, 2021-03-19:
Afaik all setpoints are single byte with resolution 1 or 0.5 (factor 1, or 2).  For flowtemps it's factor 1.
@TomoKRK do you read the setpoint or the (measured) flowtemp on your FW200?

TomoKRK, 2021-03-19:
First let me say, that I think I have to quickly jumped to the conclusion on those telegrams, while the flowSetTemp 
got changed that data in those telegrams are still the same, and I think it was coincidence that I correlated the readouts
from FW200 with the values there.

@proddy - the temperature there is also missing decimal place.

@MichaelDvP  - I am specifically referring to the values presented on my FW200. in the Info menu - there are sections
like Boiler, Heating Circuit 1, Heating Circuit 2 etc. Once I go into HC1 or HC2 - there are values for set temp with decimal places i.e. currently set temp (demanded set temp) says 38.7 for HC1 (which are radiators in my case), however this value read out in EMS says 38 only.
   I see your point re setpoints and that would make sense for me, however I am still puzzled where FW200 pulls this 
data from, or maybe it's just calculated based on the outside temperature according to the set heating curve. As I am purely 
based on the external temp controller/heating curve for my radiators & floor heating  - I don't do manual set temp points etc.

MichaelDvP, 2021-03-19:
The telegram 0x23 is new, looks like a set-telegram from thermostat to mixer. We know that from MM10, but for IPM and MM100 is not seen yet. We dont really need it, because the mixer publishes it values in the monitor message. 
In this 0x23 telegram htere is a 1A for setpoint 26°C (second 26=38°C) and a 64 for 100% pump. But the setpoint is also single byte without decimals.

Can you do a telnet session and post the output of show and the screen on the thermostat showing the value with decimals?
I dont think that Junkers uses a setpoint with 0.1 decimals and all other bosch brands do not, but we have to figure out what this value is and how to fetch it.

TomoKRK, 2021-03-19:
@MichaelDvP - I captured the log of all communication to my heating circuit 1 for several minutes. Let me know if that's
what you nedd.
`
Watching incoming telegrams, displayed in decoded format
Filtering only telegrams that match a device ID or telegram type of 0x20
000+09:41:36.973 N 4: [emsesp] Mixer(0x20) -> Boiler(0x08), ?(0x1E), data: 01 85
000+09:41:51.702 N 5: [emsesp] Mixer(0x20) -> Boiler(0x08), ?(0x1E), data: 01 85
000+09:41:59.454 N 6: [emsesp] Mixer(0x20) -> All(0x00), IPMSetMessage(0x10C), data: 01 01 00 00 00 27
000+09:42:00.124 N 7: [emsesp] Mixer(0x20) -> Boiler(0x08), UBASetPoints(0x1A), data: 27
000+09:42:00.156 N 8: [emsesp] Mixer(0x20) -> Boiler(0x08), UBASetPoints(0x1A), data: 00 (offset 2)
000+09:42:07.428 N 9: [emsesp] Mixer(0x20) -> Boiler(0x08), ?(0x1E), data: 01 85
000+09:42:08.175 N 10: [emsesp] Thermostat(0x10) -> Mixer(0x20), ?(0x23), data: 27 64 00
000+09:42:08.551 N 11: [emsesp] Thermostat(0x10) -> Mixer(0x20), ?(0x23), data: 27 64 00
000+09:42:09.495 N 12: [emsesp] Mixer(0x20) -> Boiler(0x08), UBASetPoints(0x1A), data: 27
000+09:42:20.534 N 13: [emsesp] Thermostat(0x10) <- Mixer(0x20), IPMSetMessage(0x10C), data: 01
000+09:42:20.563 N 14: [emsesp] Mixer(0x20) -> Thermostat(0x10), IPMSetMessage(0x10C), data: 01
000+09:42:20.845 N 15: [emsesp] Thermostat(0x10) <- Mixer(0x20), ?(0x10D), data: 01
000+09:42:20.870 N 16: [emsesp] Mixer(0x20) -> Thermostat(0x10), ?(0x10D), data: <empty>
000+09:42:21.971 N 17: [emsesp] Thermostat(0x10) <- Mixer(0x20), Version(0x02), data: 0A
000+09:42:22.022 N 18: [emsesp] Mixer(0x20) -> Thermostat(0x10), Version(0x02), data: 66 14 07 00 00 00 00 00 64 02
000+09:42:22.908 N 19: [emsesp] Mixer(0x20) -> Boiler(0x08), ?(0x1E), data: 01 86
000+09:42:38.135 N 20: [emsesp] Mixer(0x20) -> Boiler(0x08), ?(0x1E), data: 01 86
000+09:42:53.951 N 21: [emsesp] Mixer(0x20) -> Boiler(0x08), ?(0x1E), data: 01 86
000+09:43:00.934 N 22: [emsesp] Mixer(0x20) -> All(0x00), IPMSetMessage(0x10C), data: 01 01 00 00 00 27
000+09:43:01.669 N 23: [emsesp] Mixer(0x20) -> Boiler(0x08), UBASetPoints(0x1A), data: 27
000+09:43:01.718 N 24: [emsesp] Mixer(0x20) -> Boiler(0x08), UBASetPoints(0x1A), data: 00 (offset 2)
000+09:43:09.497 N 25: [emsesp] Mixer(0x20) -> Boiler(0x08), ?(0x1E), data: 01 86
000+09:43:21.409 N 26: [emsesp] Thermostat(0x10) <- Mixer(0x20), IPMSetMessage(0x10C), data: 01
000+09:43:21.438 N 27: [emsesp] Mixer(0x20) -> Thermostat(0x10), IPMSetMessage(0x10C), data: 01
000+09:43:21.752 N 28: [emsesp] Thermostat(0x10) <- Mixer(0x20), ?(0x10D), data: 01
000+09:43:21.778 N 29: [emsesp] Mixer(0x20) -> Thermostat(0x10), ?(0x10D), data: <empty>
000+09:43:22.877 N 30: [emsesp] Thermostat(0x10) <- Mixer(0x20), Version(0x02), data: 0A
000+09:43:22.921 N 31: [emsesp] Mixer(0x20) -> Thermostat(0x10), Version(0x02), data: 66 14 07 00 00 00 00 00 64 02
000+09:43:24.814 N 32: [emsesp] Mixer(0x20) -> Boiler(0x08), ?(0x1E), data: 01 86
000+09:43:40.539 N 33: [emsesp] Mixer(0x20) -> Boiler(0x08), ?(0x1E), data: 01 86
000+09:43:56.388 N 34: [emsesp] Mixer(0x20) -> Boiler(0x08), ?(0x1E), data: 01 86
000+09:43:56.646 E 35: [telegram] Rx: FC E0 F8 F8 F8 F8 FC FC FE FE 21 (CRC 21 != 5B)
000+09:44:02.181 E 36: [telegram] Rx: 90 0B FF 00 00 6F 03 01 (CRC 01 != BC)
000+09:44:02.450 E 37: [telegram] Rx: FC E0 FC F8 FE FF 5F (CRC 5F != 6E)
000+09:44:02.553 N 38: [emsesp] Mixer(0x20) -> All(0x00), IPMSetMessage(0x10C), data: 01 01 00 00 00 27
000+09:44:03.501 N 39: [emsesp] Mixer(0x20) -> Boiler(0x08), UBASetPoints(0x1A), data: 27
000+09:44:03.533 N 40: [emsesp] Mixer(0x20) -> Boiler(0x08), UBASetPoints(0x1A), data: 00 (offset 2)
000+09:44:09.964 N 41: [emsesp] Thermostat(0x10) -> Mixer(0x20), ?(0x23), data: 27 64 00
000+09:44:11.115 N 42: [emsesp] Mixer(0x20) -> Boiler(0x08), UBASetPoints(0x1A), data: 27
000+09:44:11.403 N 43: [emsesp] Mixer(0x20) -> Boiler(0x08), ?(0x1E), data: 01 86
000+09:44:22.103 N 44: [emsesp] Thermostat(0x10) <- Mixer(0x20), IPMSetMessage(0x10C), data: 01
000+09:44:22.133 N 45: [emsesp] Mixer(0x20) -> Thermostat(0x10), IPMSetMessage(0x10C), data: 01
000+09:44:22.414 N 46: [emsesp] Thermostat(0x10) <- Mixer(0x20), ?(0x10D), data: 01
000+09:44:22.441 N 47: [emsesp] Mixer(0x20) -> Thermostat(0x10), ?(0x10D), data: <empty>
000+09:44:23.539 N 48: [emsesp] Thermostat(0x10) <- Mixer(0x20), Version(0x02), data: 0A
000+09:44:23.583 N 49: [emsesp] Mixer(0x20) -> Thermostat(0x10), Version(0x02), data: 66 14 07 00 00 00 00 00 64 02
000+09:44:27.218 N 50: [emsesp] Mixer(0x20) -> Boiler(0x08), ?(0x1E), data: 01 86
000+09:44:42.698 N 51: [emsesp] Mixer(0x20) -> Boiler(0x08), ?(0x1E), data: 01 86
000+09:44:57.958 N 52: [emsesp] Mixer(0x20) -> Boiler(0x08), ?(0x1E), data: 01 87
000+09:45:03.936 N 53: [emsesp] Mixer(0x20) -> All(0x00), IPMSetMessage(0x10C), data: 01 01 00 00 00 27
000+09:45:04.630 N 54: [emsesp] Mixer(0x20) -> Boiler(0x08), UBASetPoints(0x1A), data: 27
000+09:45:04.662 N 55: [emsesp] Mixer(0x20) -> Boiler(0x08), UBASetPoints(0x1A), data: 00 (offset 2)
000+09:45:13.774 N 56: [emsesp] Mixer(0x20) -> Boiler(0x08), ?(0x1E), data: 01 87
000+09:45:23.165 N 57: [emsesp] Thermostat(0x10) <- Mixer(0x20), IPMSetMessage(0x10C), data: 01
000+09:45:23.197 N 58: [emsesp] Mixer(0x20) -> Thermostat(0x10), IPMSetMessage(0x10C), data: 01
000+09:45:23.509 N 59: [emsesp] Thermostat(0x10) <- Mixer(0x20), ?(0x10D), data: 01
000+09:45:23.537 N 60: [emsesp] Mixer(0x20) -> Thermostat(0x10), ?(0x10D), data: <empty>
[... cut ...]
000+09:51:28.656 N 152: [emsesp] Thermostat(0x10) <- Mixer(0x20), IPMSetMessage(0x10C), data: 01
000+09:51:28.685 N 153: [emsesp] Mixer(0x20) -> Thermostat(0x10), IPMSetMessage(0x10C), data: 01
000+09:51:28.935 N 154: [emsesp] Thermostat(0x10) <- Mixer(0x20), ?(0x10D), data: 01
000+09:51:28.960 N 155: [emsesp] Mixer(0x20) -> Thermostat(0x10), ?(0x10D), data: <empty>
000+09:51:30.084 N 156: [emsesp] Thermostat(0x10) <- Mixer(0x20), Version(0x02), data: 0A
000+09:51:30.124 E 157: [telegram] Rx: A0 10 02 00 66 14 07 00 00 00 (CRC 00 != E5)
 `

TomoKRK, 2021-03-19:
Here are the uploaded photos from FW200. Apologies but my version does not have English language supported :( 
so it's in Polish. Anyway after pressing Info button (next to Menu) you get to the "INFO FW200" section.
I highlighted the heating circuit 1 on the first pic. Once you enter this menu item, you get the details of the circuit.
The highlighted value is the "set flow temp" 39.1 ℃, translating this directly from Polish it says "Expected/Demanded/Requested Flow Temperature" - hope that helps.

[image]
[image]

MichaelDvP, 2021-03-20:
I meant the command `show` to get all values you have displayed by ems-esp, but with the log it is more clear now.

I've found the IPM manual [here](http://documents.junkers.com/download/pdf/file/6720612404.pdf) and think your system is like picture 10 (s58). The circuit hc1 is unmixed and you don't get a flowtemp value from mixer for the hc (0 in telegram 0x10C). The mixer only reads the sensor VF on the central header and publishes it in 0x1E (switch-telegram) to the boiler. The boiler republishes this value as `switchTemp` in UBAMonitorSlow (0x19) and i think this is the value you read on the FW200. The mixer setpoint is published without decimals in the mixer telegram 0x10C and send to the boiler as setpoint in 0x1A (UBASetpoints) also without decimals.  

We can add `FlowTempVf` from 0x1E to the mixer values, but it's redundant as it is republished as `switchTemp` from boiler.
The IPM mixer telegram 0x10C is wrong names, it should be `IPMStatusMessage`, i'll change this.

I did not find a english/german FW200 manual, only dutch [here](https://www.libble.eu/junkers-fw-200/online-manual-550439/?page=0042), Can you check against your polish manual which value it is and proddy can check the dutch version.
Please also check against the `switchTemp` in boiler if it is the same.

TomoKRK, 2021-03-20:
I found some FW200 user manual [here](https://www.manualscat.com/en/bosch-fw-200-manual) took a screenshot of the relevant page for Info section (attached below).
So this thread is about precision of the "Desired heating supply temperature" and it says it's calculated.

What's regarding switchTemp, in the discussion on #711 I have mentioned there that this value
corresponds to the current flow temp on HC1 (radiators). This is the same value which is described in the manual
as "Current heating supply temperature" 

[image]

proddy, 2021-03-20:
That's what I found too. 

The German manual is here: https://www.libble.eu/junkers-fw-200/online-manual-770190/

MichaelDvP, 2021-03-20:
Thanks to all, but i don't know from where the decimal comes. Is it really changing?

What i see is always the same sequenz:
`000+09:42:08.175 N 10: [emsesp] Thermostat(0x10) -> Mixer(0x20), ?(0x23), data: 27 64 00`
Thermostat tells mixer 39°C setpoint (0x27), pump on (0x64)

`000+09:43:00.934 N 22: [emsesp] Mixer(0x20) -> All(0x00), IPMSetMessage(0x10C), data: 01 01 00 00 00 27`
Mixer publishes: unmixed circuit(0x01), pump on (0x01), no valve (0x00), no measured hc flowtemperature (0x0000), setpoint 39°C (0x27)

`000+09:43:01.669 N 23: [emsesp] Mixer(0x20) -> Boiler(0x08), UBASetPoints(0x1A), data: 27`
Mixer tells boiler to set setpoint 39°C (0x27)

`000+09:43:01.718 N 24: [emsesp] Mixer(0x20) -> Boiler(0x08), UBASetPoints(0x1A), data: 00 (offset 2)`
Mixer tells boiler to switch heating off

`000+09:43:09.497 N 25: [emsesp] Mixer(0x20) -> Boiler(0x08), ?(0x1E), data: 01 86`
Mixer publishes the measured header-temperature 390 * 0.1°C (0x186) to boiler.

There are no other values and no decimals on setpoints.

TomoKRK, 2021-03-20:
@MichaelDvP - I understand the flow and it makes sense, however it does change (picture attached with the readout as of now). Please note that this also applies to heating circuit 2 (floor heating) with mixer.
   I think there is something missing in here, not sure what and where but it's very puzzling. If based on your knowledge
and experience there is nothing missing in the boiler vs thermostat vs mixer telegram exchange, and I see now that there 
are no decimal points transmitted. So where that might be coming from?
   The manual says it's calculated - I would assume it's based on the outside temp readouts and settings of heating curve,
so I would think it's calculated by thermostat (FW200) and maybe it's the only actor here holding those values while it controls other devices in the installation. However when it communicates the values behind the scenes it's always without decimal points (even for my heating circuit 2 with mixer this is still without decimal point).
   So if that's not possible to extract from transmission I am wondering if that's possible to extract from thermostat directly?

[image]

MichaelDvP, 2021-03-20:
Yes, try with `watch 10` the thermostat telegrams, maybe we find there a message containing the value. As a reference we should see the 0x23 telegram from thermstat to mixer with the integer setpoint. If this value changes, a double-byte value in another telegram should also change.

TomoKRK, 2021-03-20:
I tried to capture the comms as you identified. 
I have forced FW200 to change this value by adjusting the heating curve (moving it up and then down).

This file covers scenario of moving the curve up and the value changed from 38,6 ->39,6
[ems-log.txt](https://github.com/emsesp/EMS-ESP/files/6175495/ems-log.txt)

This log reflects the reversal so lowering curve and value changed from 39,6 -> 38,6
[ems-log-temp-down.txt](https://github.com/emsesp/EMS-ESP/files/6175496/ems-log-temp-down.txt)

MichaelDvP, 2021-03-20:
I can see the change from 38->39°C in the single byte, but there is no higher resolution value representing the decimals. Sorry. If it is a internal calculation of the thermostat, it is not published to the bus. 
On the other hand makes a setpoint with 0.1°C resolution as much sense as setting switchtimes in seconds: none. The control is much less precise, variations of some degree flowtemp does not effect the roomtemperature in short time.

@proddy do you have another idea where to search?

proddy, 2021-03-20:
not sure, we could ask Norbert. Perhaps the decimal is stored in another byte?

TomoKRK, 2021-03-20:
@MichaelDvP & @proddy - I think it's not worth to chase it further at this point in time. I thought it would be something simple and obvious which is not the case. Given all discussion I think we can close this for now and potentially reopen if something new will come up on this topic. As it really looks FW200 keeps those values and in the big picture it does not matter much whether setpoint is 39 or 39,6. 
   Thank you for assistance on it.

closed 2021-06-09 by proddy