emsesp/EMS-ESP #751: BUDERUS update time and configure timer program

Cricri66, 2021-03-21:
Hello,
First of all, my congratulations on this work.
I use a Raspberry RPI 3 as MQTT broker (Mosquitto) and I wrote android app (B4A) to control my heating (Buderus and RC30), all this works fine but I have 2 questions:
1) I cannot update buderus clock (time and date), maybe a syntax problem in my command, could you explain the correct way to do that ?
2) Will it be possible in a new ems-esp version to configure the timer program? For example, to modify the night and day changeover times.

Best regards
PS : I'm french, so excuse my english

proddy, 2021-03-21:
Welcome @Cricri66 ! We discovered recently that the RC30 doesn't allow clock setting - see https://github.com/emsesp/EMS-ESP/issues/745 

Changing the timer - I'm worried if the clock time doesn't work perhaps setting the change over times won't either.

MichaelDvP, 2021-03-21:
>  I wrote android app (B4A) to control my heating (Buderus and RC30

Nice, do you share this app? 
> to configure the timer program?

I think it's possible, but the switch times are 84 bytes, 42 switch points. It's difficult to sort this in mqtt publish and commands. You can test with raw commands. the telegrams are 0x3F for hc1, 0x49 for hc2, etc. 
Switch points are coded in 2 bytes, first byte `dayofweek<<5 + on`  dayofweek counts from 0=Mo..6=Su, 7=undefined, on is 0, 1 or 7 for undefined. Second byte is time in units of 10min counted from 00:00. A undefined switchpoint is `E7 90`

To set first hc1 switchpoint (offset 0) to thursday, on, 08:30 you have to send `call system send "0B 10 3F 00 C1 33"` 
To set second hc1 switchpoint (offset 2)  to thursday, off, 18:30 you have to send `call system send "0B 10 3F 02 C0 6F"` 
and so on.
To read the single setpoint no 10 `call system send "0B 90 3F 14 02"`
You can do the raw commands also over mqtt `system`=`{"cmd":"send","data":"0B 90 3F 14 02"}` and read answer back from `response`.

Any suggestions how to handle and name it? 
Publishing the whole program frequently is imho to much data. Maybe publish it after sending the `program` command for a single hc, like `{"hc1":{"s00":"mo,on,08:30", "s01":...,"s42":"undef"}}`
and command `{"cmd":"switchpoint","data":"01,mo,on,08:30","hc":1}` for each switchpoint?

Cricri66, 2021-03-22:
Hello,
Thank you for your reply.
Find here after my Android application (B4A) but please be indulgent, I am only an amateur programmer and I do not master all the possibilities of B4A. This program is still in development phase and I have to add other features like changing summer / winter mode and maybe others. I also have to do cosmetic modifications. For programming the timer, I will see what I can do but it is not an essential function and "timedate"doesn't work with RC30.
Today I can monitor : "seltemp", "currtemp", "outdoortemp" and "sysPress".
and modify : "daytemp", "nighttemp" and "mode"
Note that the sreen design is suitable for samsung A51 and text is in french.
Regards
[MQTT-Chauffage.zip](https://github.com/emsesp/EMS-ESP/files/6181123/MQTT-Chauffage.zip)
[image]

MichaelDvP, 2021-03-22:
@Cricri66 Thank you for the app, the B4A seems to be nice. 

I've tested a bit with my RC35 and setting the switchtimes works. There is also another program on telegrams hc1:0x42, hc2:0x4C, .., but not the actually used setpoints. (maybe circulation?). 
For esp32 (v3) i've added a `switchtime` command only with numbers as `call thermostat switchtime nn.d.o.hh:mm`, nn=number, d=day(0-6), o=on(0,1), hh:mm=time. disable switchpoint is day set to 7. Day=0 is monday. But you have to add each switchtime one by one. 
Reading/publishing the whole program is tricky because it's a lot of data. I'm waiting for a good idea.
Are you using the v3? v2(esp8266) is so much limited by memory, adding new functions is critical and if not very, very usefull, we skip.

Cricri66, 2021-03-22:
Hello,
I am using ESP8266 and I tested v2.2.2b this morning but the "Devices and Sensors" tab is not working, it is constantly searching. So, I reinstalled version v2.2.1.
My thermostat is RC30 and the time setting does not work, it is not very important the drift over 4 years of the internal clock is 10 minutes.
I continue to explore all the possibilities but will only integrate the most useful commands into my app when the modification on RC30 is tedious.

MichaelDvP, 2021-03-23:
> I tested v2.2.2b this morning but the "Devices and Sensors" tab is not working,

After update it's possible that parts in browser-cache does not fit to website data. You have to push the reload-button on your browser to make the webpage work.
> My thermostat is RC30 and the time setting does not work

Yes, proddy already wrote that we discussed this in #745. The last byte in clock setting is mostly unknown. I tested some settings with RC35 and found, that i have to set bit1 to set the clock. This works with RC35 and RC300. bit0 is daylightsavingtime, bit2,3 are flags that clock is unset, all other unknown. You can play around and test these bits.

proddy, 2021-03-23:
> Yes, proddy already wrote that we discussed this in #745. The last byte in clock setting is mostly unknown. I tested some settings with RC35 and found, that i have to set bit1 to set the clock. This [...]

@MichaelDvP shall we move the `dateTime_` register call from `Thermostat::register_device_values()` from the generic section for all thermostats back into the individual thermostat models. So that it's only active for RC35's and higher?

MichaelDvP, 2021-03-23:
@proddy Do you have feedback on which thermostats it not working? I think all thermostats have a clock. 
RC30 and RC35 are mainly identical, we have to add a section RC35 only.

RC35 - working only if bit 1 is set
RC300 - working, only tested with bit1 set, don't know if it is needed.
RC30 - not working with  bit1 set or unset, no other tests.
RC20 - unknown
Junkers - unkown
Junkers2 - unkown

I'm not really sure that RC30 is not working, maybe there are other settings in the last byte, someone with RC30 has to test some more combinations. For this byte norberts and thefisher gives different meanings.

proddy, 2021-03-23:
I tested with my RC20, with the last bit set and unset and it didn't work
```
call system send "0B 17 06 00 15 03 0B 17 26 05 01 00"
and
call system send "0B 17 06 00 15 03 0B 17 26 05 01 02"
```

MichaelDvP, 2021-03-23:
can you check first set the bit if it stays:
`call system send "0B 17 06 07 02"`
and then send the complete command.

according to norbert there are 2 bits for RTCclock
`call system send "0B 17 06 07 06"`
and
`call system send "0B 17 06 00 15 03 0B 17 26 05 01 06"`

or setting the time/date to invalid according to thefisher (possible you have to set manualy after that)
`call system send "0B 17 06 07 0C"`

Bit4 should be a running signal, worth a check with last byte `10` or `F0`.

If nothing works we should really move the command to the few working thermostats.

On RC35 the bit1 was cleared and time command does not work, then i set the bit and it worked, now it is set and it is not possible to clear it again (Maybe poweron-reset or coldstart after the goldcap is completly discharged).

MichaelDvP, 2021-03-23:
I've tested with RC35 again. Setting only a field with hour, second, etc. does not work, i have to send the complete telegram.
One single test i get a reply with bit6 also set (0x42), but gets automatically cleared next read, sending with 0x42 sets the clock, but reply is normal 02.
```
000+01:07:50.239 N 13: [emsesp] Thermostat(0x10) -> Me(0x0B), RCTime(0x06), data: 15 03 0C 17 15 30 01 02
ems-esp2:/# call system send "0B 10 06 00 15 03 0C 17 15 00 01 02"
000+01:08:30.239 N 14: [emsesp] Thermostat(0x10) -> Me(0x0B), RCTime(0x06), data: 15 03 0C 17 15 00 01 42
ems-esp2:/# read 10 06
000+01:08:41.364 N 15: [emsesp] Thermostat(0x10) -> Me(0x0B), RCTime(0x06), data: 15 03 0C 17 15 0B 01 02
ems-esp2:/#  call system send "0B 10 06 00 15 03 0C 17 15 00 01 42"
000+01:09:20.363 N 16: [emsesp] Thermostat(0x10) -> Me(0x0B), RCTime(0x06), data: 15 03 0C 17 15 00 01 02
```

proddy, 2021-03-23:
I tried all combinations, with bits set here and there and still doesn't work.

Cricri66, 2021-03-26:
Hello,
This timer problem is not important for my needs.
Thank you for your comments and suggestions and I will now leave it at that. My Android app is now sufficient to control my heating when I am away from home.
Thank you
Christian.

closed 2021-03-26 by Cricri66