emsesp/EMS-ESP #753: boiler  : "ELM LEBLANC MEGALIS CONDENS GVA C 21-5MN"

guillaume042, 2021-03-25:
hello,

For this boiler : 

- cmd "heatingactivated" does nothing. can't change the value.
- cmd "flowtemp" - new value written but as soon as the boiler start (heating or warmwater) the previous value overwrite back.

Debug Log for heatingactivated :
[heatingactivated.txt](https://github.com/emsesp/EMS-ESP/files/6207861/heatingactivated.txt)

Debug Log for flowtemp :  
show 
cmd flowtemp
opening warm water in the bathroom
show
[flowtemp.txt](https://github.com/emsesp/EMS-ESP/files/6207886/flowtemp.txt)

guillaume042, 2021-03-25:
Same thing ?
https://github.com/emsesp/EMS-ESP/issues/524#issuecomment-700980192

MichaelDvP, 2021-03-26:
I don't know your boiler, but i think it's something similar. On my boiler i have this control-panel:
[image]
It has a mechanical rotary control and if the right control is on, then heatingactivated is on. If i try setting it to off via ems, a few seconds later the controler (device_id 0x09) sends a on again. Also for flowtemp, only if the rotary is set to "auto" a flowtemp-command to the boiler is not overwritten.
You can check with `log all` the ems commands setting the value back.

guillaume042, 2021-03-26:
Here is mine : 
[image]

So i'm lost ? nothing to do ? I got to add a thermostat ?

Regards

MichaelDvP, 2021-03-27:
I think with the mechanical controls it's the same for heatingactivated, you can not send off if the knob is in on positions.
A thermostat also can not set off, it sends flowtemp 0 to switch heating off. 

A change in flowtemp for dhw is normal, the boiler sets for warm water a flowtemp of wwsetpoint+wwflowtempoffset and after preparation set the flowtemp back to heating. If you want to change the flowtemp for heating you have to send periodically, that's also what a thermostat do.

guillaume042, 2021-03-27:
> I think with the mechanical controls it's the same for heatingactivated, you can not send off if the knob is in on positions.
> A thermostat also can not set off, it sends flowtemp 0 to switch heating [...]

Thank you SO much for your precise answer !

So basicly the idea is to put flowtemp to 0 to cut the heating. As it is overwritten, i can juste check when tapewater is on, waiting for it to be back off and then putting back flowtemp to 0.

Another question if you don't mind : should i work on others parameters to optimize heating ? I've got 6 rooms with T° sensors and smart thermostatic on each heater. 

Regards

MichaelDvP, 2021-03-27:
> should i work on others parameters to optimize heating ? I've got 6 rooms with T° sensors and smart thermostatic on each heater.

I think there is no correct answer to this. The thermostats normally do
a) measure outdoortemperature and calculate flowtemp
or b) measure roomtemperature in one reference room (the one with the highest heat requirement) with all radiator/floor-thermostats fully open in this room, control flowtemp to have this room at setpoint, all other rooms are throttled by the radiator/floor-thermostats.
or c) a mixture of both modes, calculate flowtemp as a), but correct it a bit if reference room gets to cold/hot. 

All methods are not optimal. For instance with b) if sun shines in the windows of your reference room, the other rooms cool down to much. Maybe you can create a algorithm with all your smart-thermostats to check all positions and control the flow temp the way that always only one radiator/floor-thermstat is fully open. But remember that heating control is slow and roomtemperature reaction to flowtemp-change is much later.

proddy, 2021-04-06:
closing. please re-open if you have further questions or need help

closed 2021-04-06 by proddy