emsesp/EMS-ESP #755: Doc: schematic link is down

realthk, 2021-03-30:
The build your own board link is dead on the [Getting started page](https://emsesp.github.io/docs/#/getting_started).

proddy, 2021-03-31:
Hi @realthk . I removed the board designs. They were old and outdated. I'll clean up the dead links. thanks for pointing it out

realthk, 2021-03-31:
I see, but it was still a perfectly usable design, I've built mine based on the schematics found here last autumn and the only unclear thing on it was the polarity of a capacitor. 
Someone asked me on a local forum recently how to build one, directed him here, that's how I've learned about the missing schematics.
Could you please help with those old files? I was silly enough not to save back then and now cannot find with googling for "ems esp schematic"

proddy, 2021-03-31:
it's still in GitHub's history: https://github.com/emsesp/docs/blob/59f11366cb2fbcd65aed874901107c8ecc701b99/_media/schematics/Schematic_EMS-ESP.png