emsesp/EMS-ESP #757: setting flowtemp on boiler does not work in 2.2.1

mglatz, 2021-04-08:
**Bug description**
my boiler will not respond to [ip]/api?device=boiler&cmd=flowtemp&data=50 in version 2.2.1, works fine on 2.2.0

**Steps to reproduce**
call [ip]/api?device=boiler&cmd=flowtemp&data=50

**Expected behavior**
see mqtt topic boiler_data/selflowtemp = 50. Have the boiler heat the water to 50°C 

**Device information**
```
{
  "System": {
    "version": "2.2.0",
    "uptime": "000+00:18:31.299",
    "freemem": 43,
    "fragmem": 2
  },
  "Settings": {
    "enabled": "1",
    "publish_time_boiler": 0,
    "publish_time_thermostat": 0,
    "publish_time_solar": 0,
    "publish_time_mixer": 10,
    "publish_time_other": 0,
    "publish_time_sensor": 0,
    "mqtt_format": 3,
    "mqtt_qos": 0,
    "mqtt_retain": "0",
    "tx_mode": 1,
    "ems_bus_id": 11,
    "master_thermostat": 0,
    "rx_gpio": 13,
    "tx_gpio": 15,
    "dallas_gpio": 14,
    "dallas_parasite": "0",
    "led_gpio": 2,
    "hide_led": "0",
    "api_enabled": "1",
    "bool_format": 3,
    "analog_enabled": "0"
  },
  "Status": {
    "bus": "connected",
    "bus protocol": "HT3",
    "#telegrams received": 461,
    "#read requests sent": 133,
    "#write requests sent": 41,
    "#incomplete telegrams": 0,
    "#tx fails": 3,
    "rx line quality": 100,
    "tx line quality": 100,
    "#MQTT publish fails": 0,
    "#dallas sensors": 0
  },
  "Devices": [
    {
      "type": "Boiler",
      "name": "Condens 2500/Logamax/Logomatic/Cerapur Top/Greenstar/Generic HT3 (DeviceID:0x08, ProductID:95, Version:18.13)",
      "handlers": "0x10 0x11 0x14 0x15 0x16 0x18 0x19 0x1A 0x1C 0x2A 0x33 0x34 0x35 0xD1 0xE3 0xE4 0xE5 0xE6 0xE9 0xEA 0x494 0x495"
    },
    {
      "type": "Controller",
      "name": "HT3 (DeviceID:0x09, ProductID:95, Version:18.13)",
      "handlers": ""
    }
  ]
}
```
**Additional context**
I downgraded to 2.2.0 because I need this to work :). I have Junkers cerapur boiler without termostat and I directly control warm water temerature with wwsettemp (wwtemp in 2.2.0) command and heating water temperature with flowtemp. For these to work my boiler requires both the control wheels to be turned all the way to the right. Furthermore the boiler will reset these values back to wheel state (=max) every 3 minutes or so. So I have to send wwsettemp and flowtemp periodically. 
This setup works in 2.2.0 and older firmwares I used, but no longer in 2.2.1 setting flowtemp does not work and the actual flowtemp is equal to hw wheel setting, thus 88°C. wwsettemp command works in 2.2.1.

MichaelDvP, 2021-04-09:
Ups, seems that copy&paste gone wrong in [this commit](https://github.com/emsesp/EMS-ESP/commit/976474877e42b80532f560fb76fa444f3d9173b2) and sets warm water instead of flowtemp. Now fixed in the beta.

proddy, 2021-06-09:
will merge later

closed 2021-06-09 by proddy