emsesp/EMS-ESP #760: Worcestor II RF Support

stewiem2000, 2021-04-16:
Hi,
I have connected my EMS-ESP (an awesome EMS Bus Gateway Premium II (P2) from @bbqkees!) to my Worcester Bosch Greenstar 36CDi Compact Erp combi boiler with Comfort II RF wireless controller and "CRF200S" remote.  I'm running the latest dev firmware, "2.2.2b0" via telnet, "2.2.2-dev(38a443e)" from the web-interface.

The boiler seems to be detected OK (I think, certainly getting sensible values out of it), but the thermostat isn't recognised and I'm not entirely sure about the controller, _show devices_ yields
```
(1) Boiler: Worcester GBx72/Trendline/Cerapur/Greenstar Si/27i (DeviceID:0x08 ProductID:123, Version:06.02)
 This Boiler will respond to telegram type IDs: 0x10 0x11 0x14 0x15 0x1C 0x18 0x19 0x1A 0x35 0x16 0x33 0x34 0x26 0x2A 0xD1 0xE3 0xE4 0xE5 0xE6 0xE9 0xEA 

(2) Controller: Controller (DeviceID:0x09, ProductID:152, Version:03.01)

(2) Unknown: Worcester unknown (DeviceID:0x18 ProductID:216, Version:01.02)
```

_show ems_ bus information:
```
EMS Bus info:
  Tx mode: 1
  Bus protocol: HT3
  #telegrams received: 583
  #read requests sent: 132
  #write requests sent: 0
  #incomplete telegrams: 19
  #tx fails (after 3 retries): 0
  Rx line quality: 100%
  Tx line quality: 100%
```

From the following log, I'm slightly concerned by the CRC-mismatches, but I can see that dev:0x18 is the thermostat because adjusting its target temperature emits 0x2AF IDs where the third octet (value 0x20) seems to be the call for heat (value 0x21 when wanted, 0x20 when not) and the the last octet (value 0x25) seems to be twice the selected temperature, ie. 0x25 = 37 = 18.0ºC:
```
000+00:14:57.219 E 58: [telegram] Rx: 0C 2A 00 00 00 00 00 1B 00 00 00 FA 00 00 80 00 00 80 00 80 00 80 00 00 6B (CRC 6B != 62)
000+00:15:00.450 E 59: [telegram] Rx: 88 0B 19 00 80 00 00 DB 80 00 00 00 00 00 00 BC F7 04 3C 8E 00 00 00 04 7E 87 00 55 08 80 00 76 (CRC 76 != B8)
000+00:15:22.821 N 60: [emsesp] Unknown(0x18) -> All(0x00), ?(0x06), data: 15 04 13 10 15 00 04 00
000+00:15:23.442 E 61: [telegram] Rx: 80 E0 FF 09 (CRC 09 != 14)
000+00:15:23.698 N 62: [emsesp] Unknown(0x18) -> All(0x00), ?(0x2A5), data: 00 C1 20 00 00 00 25
000+00:16:03.751 N 63: [emsesp] Unknown(0x18) -> All(0x00), ?(0x2A5), data: 00 C0 20 00 00 00 25
000+00:16:22.717 E 64: [telegram] Rx: 98 00 06 00 15 04 13 10 16 00 04 (CRC 04 != D7)
000+00:16:23.197 N 65: [emsesp] Unknown(0x18) -> All(0x00), ?(0x06), data: 15 04 13 10 16 00 04 00
000+00:16:23.691 E 66: [telegram] Rx: 98 00 FF 00 01 A5 00 C0 20 00 00 00 20 09 (CRC 09 != 1D)
000+00:16:23.899 N 67: [emsesp] Unknown(0x18) -> All(0x00), ?(0x2A5), data: 00 C0 20 00 00 00 25
000+00:17:22.796 N 68: [emsesp] Unknown(0x18) -> All(0x00), ?(0x06), data: 15 04 13 10 17 00 04 00
000+00:17:23.059 E 69: [telegram] Rx: 98 00 FF 00 01 A5 00 C0 20 00 00 00 (CRC 00 != 49)
000+00:17:23.499 N 70: [emsesp] Unknown(0x18) -> All(0x00), ?(0x2A5), data: 00 C0 20 00 00 00 25
```

I've yet to try sending the boiler commands.

What can I do to help resolve the CRC errors and get my thermostat supported? :)

Thanks in advance!

MichaelDvP, 2021-04-17:
The thermostat seems to be compatible with RC100, i've added it to the database, check with new [dev](https://github.com/emsesp/EMS-ESP/releases/tag/latest).

CRC errors are most common on low buspower.  If you are on buspower, maybe the RF-thermostat draws more than wired thermostats. If you have a service-jack, try powering with this (jumper change) or try USB-power (jumper removed). 

If crc-errors stays with external power, check the bus connection and cable. Try different bus-connectors and cable length (not less 25 cm).

stewiem2000, 2021-04-18:
Thanks for the quick response! That's certainly picked it up and I can see some values propagating through to Home Assistant! :)

Unfortunately, that's where the joy seems to end :-o

Whilst the "Current room temperature" seems to be reflecting the display, the "Setpoint room temperature" is always 0º.  Looking through the code, Thermostat::process_RC300Monitor() seems to be the function used.  It even notes that octed-3 can be zero and the description for octet-6 matches what I'm seeing.  Indeed, if I add the following to the end of that function, and finally figure out how to get a working build-environment, I do seem to get actual numbers:
```
if(hc->setpoint_roomTemp == 0.0) {
    changed_ |= telegram->read_value(hc->setpoint_roomTemp, 6, 1); // is * 2, force as single byte
}
```

I'd ask whether that would be a reasonable patch-request, but having also tried to set a target temperature (via Home-Assistant or 'call thermostat temp 20'), nothing happens (other than appropriate log messages and telegram-tx) — so I wonder if there's something else (perhaps more obvious to you) that mismatches?

As for the CRC-errors: yes, I will check the cable and also try the service-jack tomorrow.

Thanks.

MichaelDvP, 2021-04-19:
Seems that this thermostat is not 100% compatible to rc100 (see #249). The check with `if(hc->setpoint_roomTemp == 0.0) {` is not a good solution, sometimes (summermode, etc.) the setpoint goes to 0 (see #256) and this logic will trigger a `changed` in every telegram. 

The 02A5 for this thermostat is very short, maybe we can use the length for checking. But first we have to check how to set the temperature. Can you log a `watch 18` in terminal for more than a minute to see also the 02B9 telegram and others from thermostat. Please also change the setpoint on thermostat while watching to log the change, Post the log here,
Please also post the output of the `show` command from telnet to see what values are read.

MichaelDvP, 2021-04-26:
@stewiem2000 To make the setpoint work for reading and writing we need a feedback from you with the mentioned log.

stewiem2000, 2021-04-26:
My apologies!  $DayJob and $Life popped-up; I had also hoped to switch over to the service-jack to gain cleaner logs (fewer invalid-CRC entries).  In the mean time, here is the requested `show` and log.  I've rebooted the gateway, let it settle, then poked through the thermostat's modes, capturing approximately a minute's worth of entries between each change.

**Show:**
```
EMS-ESP version 2.2.2b0

Boiler: Worcester GBx72/Trendline/Cerapur/Greenstar Si/27i (DeviceID:0x08 ProductID:123, Version:06.02)
  Heating active: off
  Warm water/DHW active: off
  Service code: 
  Service code number: 203
  Last error: EA(229) 08.01.2020 08:23
  Selected flow temperature: 5 °C
  Burner selected max power: 0 %
  Burner current power: 0 %
  Heating pump modulation: 0 %
  Current flow temperature: 36.2 °C
  Return temperature: 34.9 °C
  Max temperature: 36.7 °C
  Gas: off
  Flame current: 0.0 uA
  Heating Pump: off
  Fan: off
  Ignition: off
  Heating activated: on
  Heating temperature setting: 60 °C
  Circuit pump modulation max power: 100 %
  Circuit pump modulation min power: 60 %
  Circuit pump delay time: 3 min
  Burner min period: 5 min
  Burner min power: 0 %
  Burner max power: 100 %
  Temperature hysteresis on: -6 °C
  Temperature hysteresis off: 4 °C
  Burner starts: 48751
  Burner active time: 193 days 21 hours 5 minutes
  Heating active time: 183 days 7 hours 12 minutes
  Boiler total uptime: 1380 days 16 hours 18 minutes
  Maintenance message: -
  Scheduled maintenance: off
  Warm water selected temperature: 51 °C
  Warm water set temperature: 0 °C
  Warm water disinfection temperature: 70 °C
  Warm water type: flow
  Warm water charging type: 3-way valve
  Warm water circulation pump available: off
  Warm water circulation pump freq: 2x3min
  Warm water circulation active: off
  Warm water current temperature (intern): 30.2 °C
  Warm water current tap water flow: 0.0 l/min
  Warm water storage temperature (intern): 30.2 °C
  Warm water activated: on
  Warm water one time charging: off
  Warm water disinfecting: off
  Warm water charging: off
  Warm water recharging: off
  Warm water temperature ok: on
  Warm water active: off
  Warm water heating: on
  Warm water starts: 26830
  Warm water active time: 10 days 13 hours 53 minutes

Thermostat: Worcester CRF200S (DeviceID:0x18 ProductID:216, Version:01.02)
  Time: 13:26:00 26/04/2021
  (hc 1) Setpoint room temperature: 16.5 °C
  (hc 1) Current room temperature: 18.5 °C
  (hc 1) Target flow temperature: 0 °C
  (hc 1) Mode: auto

Controller: Controller (DeviceID:0x09, ProductID:152, Version:03.01)
[... 177 lines cut ...]
000+00:07:05.786 N 185: [emsesp] Thermostat(0x18) -> Me(0x0B), RC300Monitor(0x2A5), data: 00 B9 20 00 00 00 20
000+00:07:22.390 N 186: [emsesp] Boiler(0x08) -> All(0x00), UBAMonitorFast(0x18), data: 05 01 71 00 00 00 02 20 D0 01 42 80 00 01 6A 00 00 FF 00 00 00 CB 00 00 00

-----8<----- Mode changed back to Auto -----8<-----
000+00:07:30.815 N 187: [emsesp] Boiler(0x08) -> All(0x00), UBAMonitorFast(0x18), data: 05 01 73 00 00 00 02 20 D0 01 41 80 00 01 6B 00 00 FF 00 00 00 CB 00 00 00
000+00:07:31.619 N 188: [emsesp] Thermostat(0x18) -> All(0x00), RC300Monitor(0x2A5), data: 00 B9 20 00 00 00 22
000+00:07:31.814 N 189: [emsesp] Thermostat(0x18) -> Boiler(0x08), UBASetPoints(0x1A), data: 00 00
000+00:07:31.890 N 190: [emsesp] Boiler(0x08) -> All(0x00), UBAMonitorFast(0x18), data: 05 01 73 00 00 00 02 20 D0 01 41 80 00 01 6B 00 00 FF 00 00 00 CB 00 00 00
000+00:07:42.115 N 191: [emsesp] Boiler(0x08) -> All(0x00), UBAMonitorFast(0x18), data: 05 01 74 00 00 00 02 20 D0 01 40 80 00 01 6C 00 00 FF 00 00 00 CB 00 00 00
000+00:07:50.643 N 192: [emsesp] Thermostat(0x18) -> All(0x00), RC300Monitor(0x2A5), data: 00 B9 20 00 00 00 21
000+00:07:50.839 N 193: [emsesp] Thermostat(0x18) -> Boiler(0x08), UBASetPoints(0x1A), data: 00 00
000+00:08:01.742 E 194: [telegram] Rx: 88 0B 16 00 FF 3C 64 00 04 FA 05 01 03 64 3C 04 00 00 00 00 00 00 00 00 F8 00 00 00 3C 00 1E 4C (CRC 4C != 50)
000+00:08:04.140 N 195: [emsesp] Thermostat(0x18) -> All(0x00), RCTime(0x06), data: 15 04 0D 1A 16 00 00 00
000+00:08:04.331 N 196: [emsesp] Thermostat(0x18) <- Boiler(0x08), Version(0x02), data: 0A
000+00:08:04.348 N 197: [emsesp] Boiler(0x08) -> Thermostat(0x18), Version(0x02), data: 7B 06 02 00 00 00 00 00 00 0B
000+00:08:04.383 N 198: [emsesp] Thermostat(0x18) -> All(0x00), RC300Monitor(0x2A5), data: 00 B9 20 00 00 00 21
000+00:08:04.577 N 199: [emsesp] Thermostat(0x18) -> Boiler(0x08), UBAFlags(0x35), data: 01
000+00:08:04.604 N 200: [emsesp] Thermostat(0x18) -> Boiler(0x08), UBASetPoints(0x1A), data: 00 00
000+00:08:05.313 N 201: [emsesp] Thermostat(0x18) -> Me(0x0B), RC300Monitor(0x2A5), data: 00 B9 20 00 00 00 21
000+00:08:11.890 N 202: [emsesp] Boiler(0x08) -> All(0x00), UBAMonitorFast(0x18), data: 05 01 74 00 00 00 02 20 D0 01 3F 80 00 01 6A 00 00 FF 00 00 00 CB 00 00 00
000+00:08:29.864 N 203: [emsesp] Boiler(0x08) -> All(0x00), UBAMonitorFast(0x18), data: 05 01 71 00 00 00 02 20 D0 01 3E 80 00 01 67 00 00 FF 00 00 00 CB 00 00 00
000+00:08:31.839 N 204: [emsesp] Boiler(0x08) -> All(0x00), UBAMonitorFast(0x18), data: 05 01 71 00 00 00 02 20 D0 01 3C 80 00 01 67 00 00 FF 00 00 00 CB 00 00 00
```

Thanks.

MichaelDvP, 2021-04-26:
Ok, this is a new kind of thermostat, it does not support the telgrams RC300Set(0x2B9), RC300Summer(0x2AF), and RC300Curves(0x29B) Only the RC300Monitor(0x2A5) is used, but with different meaning of the values.

It's only a few bytes: `00 B8 21 00 5A 09 25` where 
pos 0/1: current roomtemp in 0.1°C
pos 2 seems to be mode bits, but is unclear
pos 3 ? always 0
pos 4 setpoint flowtemp (5A = 90°C)
pos 5 pump power (0-100%)
pos 6 the setpoint

We have to figure out the mode and mode_type settings:
mode_type seems to be bit 0 with 0-eco(off), 1-comfort(on)
mode is somewhere in bits 4 and 5, but i don't see the logic 
0x20- auto  (modetype eco)
0x21 - auto (modetype comfort)
0x30 - off (modetype eco) 

What modes does the thermostat have? I can not find a manual in web.

For the setpoint can you try if pos 6 is writeable. In telnet 
`call system send "0B 18 FF 06 01 A5 26"`
should set the setpoint roomtemp to 19°C. 
If this works we can add it to thermostat-temp-command.

stewiem2000, 2021-04-26:
Manual:
- https://www.worcester-bosch.co.uk/professional/support/literature/greenstar-comfort-ii-rf-installation-instructions-1

Modes:
- "On"; permanently try to maintain the current set-point temperature.
- "Off"; off, but does have frost-protection, so I'd guess essentially uses a set-point of 5ºC or so.
- "Auto"; follow the programmed in set-points.

Position 6 seems unwritable :(  I tried both tx_mode 1 and 2.  Same immediate response from the thermostat.
```
ems-esp:/# call system send "0B 18 FF 06 01 A5 26"
000+02:38:19.433 N 13: [emsesp] Thermostat(0x18) -> Me(0x0B), RC300Monitor(0x2A5), data: 00 B7 20 00 00 00 21
000+02:38:25.835 N 14: [emsesp] Boiler(0x08) -> All(0x00), UBAMonitorFast(0x18), data: 05 01 30 00 00 00 02 40 D0 00 E2 80 00 01 0E 00 00 FF 00 00 00 CB 00 00 00
000+02:38:36.061 N 15: [emsesp] Boiler(0x08) -> All(0x00), UBAMonitorFast(0x18), data: 05 01 30 00 00 00 02 40 D0 00 E2 80 00 01 10 00 00 FF 00 00 00 CB 00 00 00
000+02:38:45.510 N 16: [emsesp] Boiler(0x08) -> All(0x00), UBAMonitorFast(0x18), data: 05 01 30 00 00 00 02 40 D0 00 E2 80 00 01 10 00 00 FF 00 00 00 CB 00 00 00
000+02:38:55.960 N 17: [emsesp] Boiler(0x08) -> All(0x00), UBAMonitorFast(0x18), data: 05 01 30 00 00 00 02 40 D0 00 E2 80 00 01 10 00 00 FF 00 00 00 CB 00 00 00
000+02:39:01.788 N 18: [emsesp] Thermostat(0x18) -> All(0x00), RCTime(0x06), data: 15 04 0F 1A 35 00 00 00
000+02:39:01.984 N 19: [emsesp] Thermostat(0x18) <- Boiler(0x08), Version(0x02), data: 0A
000+02:39:02.020 N 20: [emsesp] Boiler(0x08) -> Thermostat(0x18), Version(0x02), data: 7B 06 02 00 00 00 00 00 00 0B
000+02:39:02.034 E 21: [telegram] Rx: 98 00 FF 00 01 A5 (CRC A5 != 87)
000+02:39:02.490 N 22: [emsesp] Thermostat(0x18) -> All(0x00), RC300Monitor(0x2A5), data: 00 B7 20 00 00 00 21
000+02:39:02.692 N 23: [emsesp] Thermostat(0x18) -> Boiler(0x08), UBASetPoints(0x1A), data: 00 00
000+02:39:03.506 N 24: [emsesp] Thermostat(0x18) -> Me(0x0B), RC300Monitor(0x2A5), data: 00 B7 20 00 00 00 21
000+02:39:06.035 N 25: [emsesp] Boiler(0x08) -> All(0x00), UBAMonitorFast(0x18), data: 05 01 30 00 00 00 02 40 D0 00 E2 80 00 01 0E 00 00 FF 00 00 00 CB 00 00 00
000+02:39:15.560 N 26: [emsesp] Boiler(0x08) -> All(0x00), UBAMonitorFast(0x18), data: 05 01 30 00 00 00 02 40 D0 00 E2 80 00 01 0E 00 00 FF 00 00 00 CB 00 00 00
```

MichaelDvP, 2021-04-27:
> Position 6 seems unwritable

That's sad, but expected. Normally the monitor telegram is not wrtiable and there is a setting-telegram for the parameters, that is published if a parameter changed. But this thermostat only publishes the monitor, seems that there is no way to write the values. I added the known values for this thermostat as special thermostat type. But i don't know what else we can do to write a value.

@proddy, @Norberts1 You know more about the different thermostats and telegrams, any idea how to write setpoint?

stewiem2000, 2021-04-28:
I'd hope you wouldn't say that, but it fits.  Thanks for adding the values, reporting will still be of some use.

Do I have any other sensible control options, e.g. "unplugging" the thermostat competely and sending the boiler commands directly?  Is that considered a Bad Ideaᵀᴹ?

If there's any additional information I can provide, please let me know.

MichaelDvP, 2021-04-28:
> Do I have any other sensible control options, e.g. "unplugging" the thermostat competely and sending the boiler commands directly? Is that considered a Bad Ideaᵀᴹ?

Normally it is not the best idea to build a own thermostat logic, the commercial thermostats are very fail-safe and control strategy is a lot of work to rebuild. But your CRF seems very simple and it seems easy to replace it. 

What the CRF do is to measure the roomtemp and calfulate to flowtemp and burnpower and send it to the boiler. This is done in only a few telegrams:
 `Thermostat(0x18) -> All(0x00), RC300Monitor(0x2A5), data: 00 B8 21 00 5A 09 25`
The thermostat publishes it's own values only for info.
`Thermostat(0x18) -> Boiler(0x08), UBASetPoints(0x1A), data: 55 09 `
The thermostat sets the boiler flowtemp (85°C) and burnPower (9%).
`Thermostat(0x18) -> Boiler(0x08), UBAFlags(0x35), data: 01`
The thermostet tells the boiler that warm water is activated. 
Also the thermostats do not have different temperatures and mode, it simple switches on or off at programmed times.

That is all i see in this log. There must be also a setting for ww-temperature in telegram 33 or 35.
If you unplug the thermostat you can do this very easy (or better if you also measure outdoor temperature) and calcualte flowtemp, also you can change between different temperatures for day and night, etc. with a bit scripting in home automation system.
You only need the commands `boiler selflowtemp x, `boiler wwsettemp x`.
Only thing is, that we do not have a dedicated command for burnPower for now (but can be added easiely), you have to send by raw system command `send "0B 08 1A 01 xx"` to change burner power. 

The best solution is to buy a (writable) masterthermostat, placed on the boiler and use the CRF as room temperature remote thermostat. I don't know much about the worcester models, but i think it's possible.

stewiem2000, 2021-05-04:
> Only thing is, that we do not have a dedicated command for burnPower for now (but can be added easiely), you have to send by raw system command send "0B 08 1A 01 xx" to change burner power.

From further monitoring, it seems the thermostat keeps the same flowtemp (0x55 / 85ºC, or 0x00 when sending "off"), and only modulates the burnPower, so having a dedicated command would certainly aid me replicating its functionality.

> The best solution is to buy a (writable) masterthermostat, placed on the boiler and use the CRF as room temperature remote thermostat. I don't know much about the worcester models, but i think it's possible.

On a whim, over the weekend, I acquired a Comfort I RF room thermostat (a CRF100S) with the hope of being able to add it into the system and glean some extra telegrams.  Unfortunately, I could not persuade it pair with either/both the CRF200S and/nor the boiler-fitted receiver :(

So it looks like you might be right and I shall consider the options of replacing the thermostat or implementing my own, unless @proddy and/or @norberts1 have any further suggestions...?