emsesp/EMS-ESP #762: ESP8266 fails to connect with mesh wifi

VincentSC, 2021-04-20:
Seems that ESP8266 fails to connect with a wifi that is part of a mesh-network. Is there a solution, or should I upgrade the gateway to ESP32? https://bbqkees-electronics.nl/wiki/gateway/upgrade-the-gateway-to-an-esp32.html

proddy, 2021-04-21:
it's the same wifi code in both ESP8266 and ESP32 variants. What is the error exactly?

VincentSC, 2021-05-01:
Solved it. Control-channel 13 and 14 do not seem to work with the EMS-ESP.
 
[image]

closed 2021-05-01 by VincentSC

proddy, 2021-05-02:
glad you fixed it. do any of your other ESP8266 projects work with Wifi ch 13 or 14?

VincentSC, 2021-05-02:
Currently no such projects active. Too busy to hobby.

It was based on searching specifically for ESP8266 and channel 13. Found various issues on Github. Some claimed it was solved by changing the channel, others said it should work and some said it depended on the country if it was allowed. As it was a discussion point, I turned off "allow channels 13 and 14" for the control channel and it WFM (works for me), so a possible solution. It could also be the combination of mesh and channel 13, but I have not researched it.

If somebody reads this and is experimental, put the wifi router on channel 13 or 14 (howto: allow channel 13/14 and have a lot of neighbours) and see of it fails.

proddy, 2021-05-02:
yes looks like Espressif added this to the core library. https://github.com/esp8266/Arduino/issues/7083

VincentSC, 2021-05-03:
> surely a backwards breaking feature

😄