emsesp/EMS-ESP #763: tankMiddleTemp for my HT3 boiler vanished on 2.2.2b0

joanwa, 2021-04-25:
Just upgraded to 2.2.2b0 and `tankMiddleTemp` disappeared from telnet using `show values` as well as mqtt. Just saw this line:

https://github.com/emsesp/EMS-ESP/blob/473c5509e621206522dad8fa54b802dedf01cd6c/src/devices/boiler.cpp#L247

which makes it non HT3 devices only. My boiler is HT3 and it worked before. Is there a conflict with something else?

```
ems-esp:/$ show values
Boiler: HT3 (DeviceID:0x08, ProductID:95, Version:34.12)
```

MichaelDvP, 2021-04-26:
It was the first attempt to sort the messages for boiler models to free up a little memory and sort out empty values that gives HA complains. 
Can you please check the other non-ems1.0 telegrams if they contains data for your boiler:
`read 08 26`, `read 08 D1`, `read 08 E3`, `read 08 E4`, `read 08 E5`, `read 08 E6`, `read 08 E9`, `read 08 EA`
I'll add back the 2A-telegram and if there is another.

joanwa, 2021-04-26:
you mean a `watch raw 0x08`?

MichaelDvP, 2021-04-27:
No, i mean the read command for differet telegrams as written.

joanwa, 2021-04-27:
it's just 2A
```
ems-esp:/# read 08 26
ems-esp:/# read 08 D1
ems-esp:/# read 08 E3
ems-esp:/# read 08 E4
ems-esp:/# read 08 E5
ems-esp:/# read 08 E6
000+12:03:34.325 E 4: [telegram] Last Tx Read operation failed after 3 retries. Ignoring request.
ems-esp:/# read 08 E9
ems-esp:/# read 08 EA
ems-esp:/# read 08 2A
000+12:03:46.872 N 5: [emsesp] Boiler(0x08) -> Me(0x0B), ?(0x2A), data: 15 00 00 00 03 02 00 00 00 00 00 80 00 00 02 11 80 00 01 BE
```

MichaelDvP, 2021-04-27:
Ok, i've added it back, also telegram 0x26 for wwMaxPower modulation that gives no reply on your boiler. But i think it's in some modulated HT3 boilers and the HA is now checked by valid values and should not make a problem.

joanwa, 2021-04-27:
great, works again!
```
ems-esp:/# read 08 2A
000+00:13:45.490 N 7: [emsesp] Boiler(0x08) -> Me(0x0B), MC10Status(0x2A), data: 15 00 00 00 03 02 00 00 00 00 00 80 00 23 02 18 80 00 02 92
```
but not

closed 2021-05-02 by proddy