emsesp/EMS-ESP #764: mqtt payload mode and modetype is missing for thermostat_data_hc2

deejaybeam, 2021-05-02:
**Bug description**
After upgrading from v2.1.0 to v2.2.1 the attributes 'mode' and 'modetype' are missing for topic 'thermostat_data_hc2', but they are existing for 'thermostat_data_hc1' and where present before upgrading.

**Steps to reproduce**
ems-esp v2.2.1 with mqtt-setting-format single and two heating-circuits

**Expected behavior**
Added/fixed hc2-topic including values for 'mode' and 'modetype'

**Screenshots**
[image]
[image]

**Device information**
```
{
  "System": {
    "version": "2.2.1",
    "uptime": "000+00:25:28.816",
    "freemem": 38,
    "fragmem": 6
  },
  "Status": {
    "bus": "connected",
    "bus protocol": "Buderus",
    "#telegrams received": 2104,
    "#read requests sent": 550,
    "#write requests sent": 0,
    "#incomplete telegrams": 0,
    "#tx fails": 0,
    "rx line quality": 100,
    "tx line quality": 100,
    "#MQTT publish fails": 0,
    "#dallas sensors": 0,
    "#dallas fails": 0
  },
  "Devices": [
    {
      "type": "Boiler",
      "name": "GB125/MC10 (DeviceID:0x08, ProductID:72, Version:02.04)",
      "handlers": "0x10 0x11 0x14 0x15 0x16 0x18 0x19 0x1A 0x1C 0x2A 0x33 0x34 0x35 0xD1 0xE3 0xE4 0xE5 0xE6 0xE9 0xEA 0x494 0x495"
    },
    {
      "type": "Thermostat",
      "name": "RC30 (DeviceID:0x10, ProductID:67, Version:02.08)",
      "handlers": "0xA3 0x06 0xA2 0x12 0x3E 0x3D 0x3F 0x48 0x47 0x49 0x52 0x51 0x53 0x5C 0x5B 0x5D 0xA5 0x37"
    },
    {
      "type": "Mixer",
      "name": "MM10 (DeviceID:0x21, ProductID:69, Version:02.00)",
      "handlers": "0xAB"
    },
    {
      "type": "Controller",
      "name": "BC10/RFM20 (DeviceID:0x09, ProductID:68, Version:02.00)",
      "handlers": ""
    }
  ]
}
```

**Additional context**
content of the mqtt-topics:
```
thermostat_data_hc1 = {"seltemp":0,"daytemp":21,"nighttemp":14,"holidaytemp":17,"nofrosttemp":5,"heatingtype":"radiator","targetflowtemp":5,"offsettemp":0,"designtemp":60,"roominfluence":3,"flowtempoffset":5,"minflowtemp":5,"maxflowtemp":65,"summertemp":16,"reducemode":"outdoor","program":0,"mode":"auto","modetype":"night"}
thermostat_data_hc2 = {"seltemp":26.5,"currtemp":24,"daytemp":26.5,"nighttemp":10.5,"holidaytemp":17,"nofrosttemp":5,"heatingtype":"radiator","targetflowtemp":46,"offsettemp":-2,"designtemp":60}
```

MichaelDvP, 2021-05-02:
Looks to me like a memory related issue: not enough free heap to send the complete mqtt-message. 
Try to set the mqtt-publish times to longer cycles (do you have set it `on change`?).

deejaybeam, 2021-05-02:
Hi,
No, it is set to 60sec
[image]

and sys looks like this:
[image]

MichaelDvP, 2021-05-02:
10k free is ok and should work. Are the data also missing on webpage and API. Try `http://ems-esp.local/api?device=thermostat&cmd=info&id=2`

MichaelDvP, 2021-05-02:
Ah, i think i've found it. Please try with the [dev2.2.2b0](https://github.com/emsesp/EMS-ESP/releases/tag/latest)

deejaybeam, 2021-05-02:
hi, data is available via api-call.
after installing dev2.2.2b0 data is now there!
Thank you very much!

proddy, 2021-05-02:
thanks @MichaelDvP for finding that nasty bug. Shall we make official releases of both the ESP8266 and ESP32 firmwares? It's been a while

MichaelDvP, 2021-05-02:
> Shall we make official releases of both the ESP8266 and ESP32 firmwares?

Yes, for v2 esp8266 we need a bugfix-release, fixing setflowtemp, single mqtt and adapt to this nasty HA-change.

For v3 it's mainly a feature release, there are so many nice new features in the dev, and it seems to me that most people do not install the dev and waiting for the release.

BTW: i'm testing to have a v2-dev with v3-compatible namings and mqtt telegrams, makes it easier to switch between the adapters. I also try to free up more memory. The standalone-build throw errors, but on esp it works with ~4k more free heap and reload page with open telnet. Should i leave it in my repo, or create a new branch here?

proddy, 2021-05-03:
@MichaelDvP ok. I'll do an official release for 2.2.2 and bump 3.0.3 to 3.1.0 since it contains new features. I'll put in a fix for the MQTT reconnect timeout which stops after a while and then do it.
Backing porting v3 naming to v2 is quite a task so kudos if you get it all working and also save on the heap. Could you make a branch here so I can help?

closed 2021-05-04 by proddy