emsesp/EMS-ESP #765: Bosch Condens GC2300W/Logamax Plus GB122 - missing error message

partikus, 2021-05-03:
**Bug description**

Once the boiler stopped working due to an issue, the error sent is not being handled. 

```
?(0xBF), data: 08 EA 00 06 11 20 2D 2D 0B 87 00 00 00 00 00 00 00 00 00 00 00 00 00 00
```

Based on [the error message type](https://emsesp.github.io/docs/#/MC110-controller?id=type-0xbf-deviceserrormessage) I've decoded the following values:

```
# pattern
[byte:{number}] | [value] - [notes]

[b:0] | 08 - [8] device ID
[b:1] | EA - [234] - model ID
[b:2] | 00 - [0] - blocking error: always 0x0A (fault class, see service manual) 
[b:3] | 06 - [6] blocking error: 0x00 or 0x10 = interlocking error, 0x11 = blocking error, 0x12 = system error, 0x13 = reset system error
[b:4] | 11 - [DC1] - device control ASCII | Fault code 0. character (possibly 0x0a = space)
[b:5] | 20 - [ ] ASCII | Fault code (operation code) 1st character
[b:6] | 2D - [E] Fault code (operation code) 2nd character
[b:7] | 2D - [E] additional code (status code)
[b:8] | 0B - [...
[b:9] | 87 - ...] - 2951 - data 
```

```
Boiler: Bosch Logamax Plus GB122 (DeviceID:0x08 ProductID:234, Version:05.03)
 This Boiler will respond to telegram type IDs: 0x10 0x11 0x14 0x15 0x16 0x18 0x19 0x1A 0x1C 0x2A 0x33 0x34 0x35 0xD1 0xE3 0xE4 0xE5 0xE6 0xE9 0xEA
 ```

**Device information**
```json
{
  "WIFI": {
    "ssid": "XXXX",
    "hostname": "ems-esp",
    "static_ip_config": "off"
  },
  "AP": {
    "provision_mode": 1,
    "ssid": "ems-esp",
    "local_ip": "192.168.4.1",
    "gateway_ip": "192.168.4.1",
    "subnet_mask": "255.255.255.0"
  },
  "MQTT": {
    "enabled": "on",
    "host": "192.168.98.134",
    "port": 1883,
    "username": "mqtt",
    "client_id": "ems-esp",
    "keep_alive": 120,
    "clean_session": "on",
    "max_topic_length": 128,
    "publish_time_boiler": 0,
    "publish_time_thermostat": 0,
    "publish_time_solar": 0,
    "publish_time_mixer": 0,
    "publish_time_other": 5,
    "publish_time_sensor": 0,
    "mqtt_format": 3,
    "mqtt_qos": 0,
    "mqtt_retain": "on"
  },
  "NTP": {
    "enabled": "on",
    "server": "0.pl.pool.ntp.org",
    "tz_label": "Europe/Warsaw",
    "tz_format": "CET-1CEST,M3.5.0,M10.5.0/3"
  },
  "OTA": {
    "enabled": "on",
    "port": 8266
  },
  "Settings": {
    "tx_mode": 2,
    "ems_bus_id": 11,
    "syslog_level": -1,
    "syslog_mark_interval": 0,
    "syslog_host": "",
    "master_thermostat": 0,
    "shower_timer": "off",
    "shower_alert": "off",
    "rx_gpio": 13,
    "tx_gpio": 15,
    "dallas_gpio": 14,
    "dallas_parasite": "off",
    "led_gpio": 2,
    "hide_led": "on",
    "api_enabled": "on",
    "bool_format": 1,
    "analog_enabled": "off"
  }
}
```

MichaelDvP, 2021-05-04:
Is there a special reason why adding this message? In my understanding the devices publishes on error the 0xBE/0xBF messages, the master collects them and republishes service-code/service-message. Also if pos 3 in 0xBF is set, the error is logged to 0x10, 0x11 or 0x12. We already read service-code and the logs. I think the information in BE/BF is redundant and to read it we need to register this telegrams to every device, wasting a lot of memory in poor 8266.
But maybe my understanding is wrong, luckily i have not much experience with errors yet.

partikus, 2021-05-04:
@MichaelDvP thanks for the update. 

Good to know the software ignores some messages due to the limitations of esp8266. I do not want to force adding new messages but would be more than cool if all messages (including unsupported as well) could be sent to MQTT.

**The problem:** I'd like to send a push/email notification through HA if the boiler throws an error like 2951.

Could you advice me how reading unsupported telegrams can be achieved?

MichaelDvP, 2021-05-04:
> Good to know the software ignores some messages due to the limitations of esp8266. I do not want to force adding new messages 

A lot of messages are not read. Some because there is no new information, much more because the fields are unknown. Also in known messages not all fields can be interpreted. . 
 
Dont you get the error as status-code? What does error 2951 stands for?

You can send a raw message to the bus and read the response via mqtt. Send to topic system:
`{"cmd":"send","data":"0b 88 bf 00 20"}`
and see the `response` topic. 
If you only want to read the errorcode use  `{"cmd":"send","data":"0b 88 bf 08 02"}`

partikus, 2021-05-05:
Could you explain the data payload, please? `0b 88 bf 08 02`

MichaelDvP, 2021-05-05:
> Could you explain the data payload, please? 0b 88 bf 08 02

It's from emsesp `0b` to boiler `08` as read `| 80` = `88`, message `bf`, offset `08` 2 bytes `02`

proddy, 2021-09-19:
@partikus from the discussion on Discord, if you've moved over to V3 I can transfer this issue to the the EMS-ESP32 project and take another look

partikus, 2021-09-20:
After serviceman visit I realized that last 10 errors are saved. So I've started looking for data I've mentioned and found this:

```
read 08 C2 08
```

Then got:

```
004+04:58:33.375 N 22: [emsesp] Boiler(0x08) -> Me(0x0B), ?(0xC2), data: 0B 84 00 0F 3D 6A 00 00 0F 3D 6B 00 08 EA 00 0A 09 20 2D 2D 0B 84 00 0F 2F 87 00 (offset 8)
004+04:58:34.436 N 23: [emsesp] Boiler(0x08) -> Me(0x0B), ?(0xC2), data: 87 00 00 0F 2F 88 00 08 EA 00 0A 09 20 2D 2D 0B 6F 00 0F 2F 07 00 00 0F 2F 08 00 (offset 33)
004+04:58:34.586 N 24: [emsesp] Boiler(0x08) -> Me(0x0B), ?(0xC2), data: 08 00 08 EA 00 0A 09 20 2D 2D 0B 84 00 0F 2F 06 00 00 0F 2F 07 00 08 EA 00 0A 09 (offset 58)
004+04:58:34.776 N 25: [emsesp] Boiler(0x08) -> Me(0x0B), ?(0xC2), data: 0A 09 20 2D 2D 0B 86 00 0F 1A F2 00 00 0F 1A F3 00 08 EA 00 0A 09 20 2D 2D 0B 84 (offset 83)
004+04:58:35.026 N 26: [emsesp] Boiler(0x08) -> Me(0x0B), ?(0xC2), data: 0B 84 00 0F 1A EA 00 00 0F 1A EA 00 08 EA 00 0A 09 20 2D 2D 0B 6F 00 0F 18 CB 00 (offset 108)
004+04:58:35.177 N 27: [emsesp] Boiler(0x08) -> Me(0x0B), ?(0xC2), data: CB 00 00 0F 18 CC 00 08 EA 00 0A 09 20 2D 2D 0B 84 00 0F 18 CB 00 00 0F 18 CB 00 (offset 133)
004+04:58:35.367 N 28: [emsesp] Boiler(0x08) -> Me(0x0B), ?(0xC2), data: CB 00 08 EA 00 0A 09 20 2D 2D 0B 6F 00 0F 13 06 00 00 0F 13 06 00 08 EA 00 0A 09 (offset 158)
004+04:58:35.787 N 29: [emsesp] Boiler(0x08) -> Me(0x0B), ?(0xC2), data: 0A 09 20 2D 2D 0B 84 00 0F 13 05 00 00 0F 13 06 00 (offset 183)
```
After formatting, the 2 bytes after `08 EA 00 0A 09 20 2D 2D` looks like the errors:

```
0B 84 00 0F 3D 6A 00 00 0F 3D 6B 00 
0B 84 00 0F 2F 87 00 87 00 00 0F 2F 88 00 
0B 6F 00 0F 2F 07 00 00 0F 2F 08 00 08 00 
0B 84 00 0F 2F 06 00 00 0F 2F 07 00 08  
0B 86 00 0F 1A F2 00 00 0F 1A F3 00 
0B 84 00 0F 1A EA 00 00 0F 1A EA 00 
0B 6F 00 0F 18 CB 00 CB 00 00 0F 18 CC 00 
0B 84 00 0F 18 CB 00 00 0F 18 CB 00 CB 00 
0B 6F 00 0F 13 06 00 00 0F 13 06 00 08 
0B 84 00 0F 13 05 00 00 0F 13 06 0
```

your help is more than welcome

proddy, 2021-09-20:
the v2 (ESP8266 version) is discontinued so moving this over to the EMS-ESP32 project as a question and possible enhancement if there is a use case that's useful for others. Note it would mean upgrading to v3.