emsesp/EMS-ESP #766: V2.2.2 Selected flow temperature is missing

KNXBroker, 2021-05-11:
**Bug description**
Since V2.2.2 Selected flow temperatur is missing in Home Assistant. 

**Steps to reproduce**
Home Assistant 2021.5.2
EMS-ESP V2.2.2
Boiler HT3
Device ID 0x08
Product ID 95
Version 18.11

**Expected behavior**
Selected flow temperatur should be available in HomeAssistant. 

**Device information**
```
{
  "System": {
    "version": "2.2.2",
    "uptime": "000+00:07:38.403",
    "freemem": 42,
    "fragmem": 12
  },
  "Status": {
    "bus": "connected",
    "bus protocol": "HT3",
    "#telegrams received": 244,
    "#read requests sent": 74,
    "#write requests sent": 0,
    "#incomplete telegrams": 21,
    "#tx fails": 1,
    "rx line quality": 92,
    "tx line quality": 98,
    "#MQTT publish fails": 0,
    "#dallas sensors": 0,
    "#dallas fails": 0
  },
  "Devices": [
    {
      "type": "Boiler",
      "name": "HT3 (DeviceID:0x08, ProductID:95, Version:18.11)",
      "handlers": "0x10 0x11 0x14 0x15 0x1C 0x18 0x19 0x1A 0x35 0x16 0x33 0x34 0x26 0x2A"
    },
    {
      "type": "Thermostat",
      "name": "FW120 (DeviceID:0x10, ProductID:192, Version:53.02)",
      "handlers": "0xA3 0x06 0xA2 0x12 0x16F 0x170 0x171 0x172 0x165 0x166 0x167 0x168"
    },
    {
      "type": "Controller",
      "name": "HT3 (DeviceID:0x09, ProductID:95, Version:18.11)",
      "handlers": ""
    }
  ]
}
```

**Additional context**
I assume it is due to the new filtering functionality.

proddy, 2021-05-12:
is it only missing in HA?
Can you see it in the Web and Console?
Is it in the MQTT topic called `boiler`?
What is in the MQTT topic called `homeassistant/sensor/ems-esp/boiler_selflowtemp`?
Was it there in 2.2.1?

MichaelDvP, 2021-05-12:
It's a typo in boiler.cpp line 136. I'll fix the dev.

commit c56dc7b3 (2021-05-12): fix #766 selFlowTemp in HA
closed 2021-06-09 by proddy