emsesp/EMS-ESP #768: Thermostat off mode - CW100/RC200

Flavio5, 2021-05-26:
Hi, I've update the firmware from 2.1.0 to 2.2.2 (esp8266), i've a CW100 thermostat and cerapur modul solar boiler.
All works well but the climate entity in home assistant doesn't select off mode for heating circuit (the choises are: heat, auto, off), in the previous firmware version it works.
I think there is a bug in mqtt pubblish payload: i have try to pubblish "off" in the topic "ems-esp/thermostat_hc1" and nothing happend, so the right version is " 'off' " (without spaces) and it works (the thermostat selects heat mode e a temperature of 0* degrees).

It is a bug firmware or it is possibile to solve from home assistant?

proddy, 2021-06-09:
I think this is the same as https://github.com/emsesp/EMS-ESP32/issues/66 which was fixed in the v3 branch (ESP32). I'll take a look

proddy, 2021-07-17:
There have been so many new fixes and enhancements since v2 so I would strongly recommend you switch to v3 and swap out the ESP8266 for an ESP32. Porting the fixes back to v2 is going to be too complicated I'm afraid.

closed 2021-07-17 by proddy