emsesp/EMS-ESP #769: ems-esp might not reconnect to mqtt in some situations, after connection is lost

hamster65, 2021-06-03:
Hi,

my raspberry was not reachable and I had to to power cycle it. Two days later, I noticed that data was not logged from the boiler. EMS-ESP showed that the connection to the broker was lost (I think it said "TCP error"). I rebooted ems-esp from the web interface and it worked again.

I'd have expected it to automatically reconnect to the broker... and it does, when I stop the mqtt service. So it seems that in some situations, this does not work.
I also found https://github.com/emsesp/EMS-ESP/issues/705 where a similar observation was reported.

proddy, 2021-06-03:
Yes this is a known problem in the Async and MQTT library. See https://github.com/emsesp/EMS-ESP32/issues/42 

The code needs changing to implement a more basic try & re-try mechanism.

hamster65, 2021-06-03:
Ah, I didn't see this issue. Thanks!

closed 2021-06-03 by hamster65