emsesp/EMS-ESP #770: Guaranteed change via MQTT

swa72, 2021-06-13:
I send requests via MQTT from HA to EMS-ESP like changing the offset temp of a circuit. Sometimes the EMS-ESP does not change anything. Anyone an idea of a "guaranteed" change "concept" like

count = 0

repeat until sensor = value x or counter = 5
- increase counter
- send request via MQTT
- wait 30s
- read value of sensor

if sensor != value x then send message "error"

-Stefan.

MichaelDvP, 2021-06-14:
If you dont have any tx-error, the mqtt message is lost. Set mqtt-qos to 1, than the message is resent until received.
[see here](https://en.wikipedia.org/wiki/MQTT#Quality_of_service_(QoS))

proddy, 2021-10-04:
Recommend using the REST API to have HA send commands as the API will return a response code

closed 2021-10-04 by proddy