emsesp/EMS-ESP #772: Feature Request: Description for Dallas sensors

alexandergeiler, 2021-07-13:
I have 9 Dallas sensors connected to my EMS-ESP.
It is hard to distinguish them because you only see the ID and it is hard to remember IDs.

I would like to have a description field for each sensor (see screenshot)
[image]
This will make it much easier to identify which sensor is taking which readings

I would be great to have the description also available at MQTT (see screenshot)
[image]

MichaelDvP, 2021-07-14:
You know that v2 for esp8266 will not get any new enhancements, the 8266 is to memory limited and with some more ems-devices like your system access to webpage is critical. Better switch to [esp32](https://github.com/emsesp/EMS-ESP32) and v3 with more features.

We had discussed this before, the id is not very intuitive and the number can change, an alias would help. I've made an approach to replace the id by a name in all fields (web/mqtt/HA/terminal) and also add an offset calibration to sensors. But for now there is only a terminal command to set the name `sensor [id] [name] [offset]`, `sensor` list the names, `sensor  id` clears an id-entry, `sensor id name` set a name, `sensor id name offset` sets name and offset, name can be `""` to leave it as id. The sensor-format selection is enhanced to `by number/by id/by name` 
I'm thinking about an easier way to set the names, so it is not in the official build, only in my [test](https://github.com/MichaelDvP/EMS-ESP32/releases).

alexandergeiler, 2021-07-16:
Ohhh, I posted it in the wrong project. I have a esc32 :-)
Don´t know how to move that feature request in the correct project...

Nice to hear that there is already an idea to implement something.
then I'll wait until this is available as an official feature.

MichaelDvP, 2021-07-16:
I've integrated it in the official [v3.1.2b4](https://github.com/emsesp/EMS-ESP32/releases). In terminal use `sensorname` command and you can switch in web-Settings the `Dallas Format` to `by Name`, affects mqtt, api and web.

> Don´t know how to move that feature request in the correct project...

Close this and reopen in eps32 if it does not work as expected. I also don't know a way to move.

proddy, 2021-07-16:
I can move it when I'm back this weekend!