emsesp/EMS-ESP #773: MQTT publishes non-populated hc(x) configs in home-assistant mode leading to spammed error and warning logs in HA

joanwa, 2021-08-29:
**Bug description**
*A clear and concise description of what the bug is. Mention which EMS-ESP version you're using.*
*using v2.2.3b0
* MQTT publishes non-populated hc(x) configs in home-assistant mode leading to avalanches of error and warning logs in HA
* my example: I have one heating circuit. telnet `show values` gives the values only for heating circuit 1
* over mqtt, addionally to `hc1`, I get config messages for `hc2`, `hc3`, ... 
* example for a `hc2` config topic: `homeassistant/sensor/ems-esp/mixer_hc2_flowSetTemp/config` 
```
{
"name":"Mixer hc2 Setpoint flow temperature",
"uniq_id":"mixer_hc2_flowSetTemp",
"unit_of_meas":"°C",
"stat_t":"ems-esp/mixer_data",
"val_tpl":"{{value_json.hc2.flowSetTemp}}",
"ic":"mdi:coolant-temperature",
"dev":{"ids":["ems-esp-mixer"]}
}

```
but `ems-esp/mixer_data`
```
{
"hc1":{"type":"hc","flowSetTemp":45,"pumpStatus":"on","valveStatus":0}
}
```
obviously only populates values for `hc1`, since I don't have values for `hc2`, `hc3` ...
Therefore home-assistant mqtt subscriber throws on every new published message the following errors:

```
Logger: homeassistant.helpers.template
Source: helpers/template.py:1417
First occurred: 1:23:53 PM (5154 occurrences)
Last logged: 2:14:21 PM

Template variable error: 'dict object' has no attribute 'wwc2' when rendering '{{value_json.wwc2.pumpStatus}}'
Template variable error: 'dict object' has no attribute 'wwc2' when rendering '{{value_json.wwc2.tempStatus}}'
Template variable error: 'dict object' has no attribute 'wwc1' when rendering '{{value_json.wwc1.wwTemp}}'
Template variable error: 'dict object' has no attribute 'wwc1' when rendering '{{value_json.wwc1.pumpStatus}}'
Template variable error: 'dict object' has no attribute 'wwc1' when rendering '{{value_json.wwc1.tempStatus}}'
```

**Steps to reproduce**
IMO everyone using home assistant with less than all possible heating circuits used should get those error messages.

**Expected behavior**
* only publish heating circuit config topics that ems-esp actually found.

proddy, 2021-08-29:
This is caused in a change Home Assistant made to template validation with Discovery. It was fixed in v3 (https://github.com/emsesp/EMS-ESP32/issues/56) a while back.

Note I'm no longer supporting the ESP8266 (this repo). It's all ESP32 now and you'll find many new additions. Just swap out your ESP8266 with an ESP32 (it costs like 4 euros). See https://bbqkees-electronics.nl/wiki/gateway/upgrade-the-gateway-to-an-esp32.html

joanwa, 2021-08-29:
thanks for the info. I can't change until there's an ESP that also runs on bus power :/

proddy, 2021-08-29:
oh yeah. I remember. @bbqkees any ideas?

joanwa, 2021-08-29:
hm, this is more or less a duplicate of https://github.com/emsesp/EMS-ESP/issues/756

bbqkees, 2021-08-30:
From my side there will never be an ESP upgrade for the Premium II that will work with bus power.
I also have no new product planned that will work in bus powered mode.
With the feature-rich and demanding EMS-ESP32 combined with all those power hungry new thermostat models the chance that it will work in any customer situation is just not big enough to make it available as a general feature. 

All ESP32 Gateway products are very solid performers on the bus when used with the DC power supply or service jack. Adding a bus powered mode would be quite detrimental in a lot of situations because of the low power budget on the bus.

All boilers need a mains socket nearby anyway so there you can also plug in a USB adapter for a Premium II upgrade.

joanwa, 2021-08-30:
ok, I thought in the direction of ESP32-S2. There seems to be quite an upside potential to power efficiency, see here: https://diyi0t.com/reduce-the-esp32-power-consumption/

bbqkees, 2021-08-30:
Sleep modes of the ESP32 and ESP32-S2 are great for overall power reduction in battery powered systems.
Benefit of the S2 only really comes into play when deep sleep is active.
However, most power reduction in the sleep modes comes from powering off the WiFi, which should be active all the time with EMS-ESP (you cannot receive commands when WiFi is off).

Also when the ESP32 is not in sleep mode (with WiFi on), it still draws a lot of current at those times which is not a problem for the battery but it is for the EMS bus.
And when you setup EMS-ESP the AP is on, which really uses a huge amount of power. It has to come from somewhere.

MichaelDvP, 2021-08-30:
1. I think the mqtt publish is only for active hc, maybe this false hc are stored (retain) in mqtt broker. Try to delete all from mqtt broker and reconnect.
2. in v3 there are options to reduce tx-power and clock down the esp32-core, BT is also disabled now. I think noone have tried with buspower and all energy-saving options yet.
3. Why is usb-power not possible. All modules like MM100, etc have a second 230V clamp, buy a [europlug](https://www.ebay.de/itm/203577063036?hash=item2f6623767c:g:JCQAAOSwWNNdrDw3), connect it to the clamp and plugin a small usb-psu.

closed 2021-10-04 by proddy