emsesp/EMS-ESP #774: EMS-ESP Thermostat is not recognized/added to homeassistant lovelace autodiscover

bartplessers, 2021-08-30:
Hi,

Running

- BUDERUS Logamax plus GB192iT40
- BOSCH RC310 thermostaat
- BUDERUS Gateway Logamatic web KM100
- BBQkees Gateway Premium II
- ems-esp v2.2.2
- home assistant 2021.8.8

The thermostat is discovered with it's according entities by homeassistant:
[image]

However, in the past, this thermostat was automatically added to the lovelace interface of homeassistant.
Recently, I reinstalled my instance of homeassistant, and I notice that the thermostat does **not** show up anymore.

Also, If I add the entities manually to the dashboard, the thermostat seems "unavailable":
[image]

On MQTT Explorer, I see data coming in:
[image]

Question:
How can I add the thermostat to my homeassistant dashboard.

kind regards,
Bart Plessers

proddy, 2021-08-30:
go into MQTTExplorer and see if the Discovery topics are there. Search for `hc1`. This is what I see:

[image]

Also I'm no longer supporting the ESP8266 so I would strongly recommend swapping it out for an ESP32 and jumping over to v3. You're missing out on a lot of cool new features.

bartplessers, 2021-08-30:
Hi @proddy ,

I do have (more or less) the 'sensor' topics:
[image]

but I'm missing the 'climate' topic:
[image]

That's probably the reason why HA does not recognize the thermostat...

Any idea why?

>Also I'm no longer supporting the ESP8266 so I would strongly recommend swapping it out for an ESP32 and jumping over to v3. You're missing out on a lot of cool new features.

I will give it a try in the future, but I recently (+/- 1Year ago) bought a BBQkees Gateway Premium II (P2), so If I get this working, it's fine for nox.

grtz 
B

proddy, 2021-09-01:
I'm not sure what is causing this, assuming you hadn't upgrade the firmware or changed the settings (like tx_delay). I would again strongly recommend you buy an ESP32 for 4 euros and swap it out on your Premium II board.

bbqkees, 2021-09-01:
What sometimes does help is removing the Gateway from power/ bus completely and then connect it again after 20 seconds.
Sometimes rebooting via the web is not enough.

bartplessers, 2021-09-02:
@bbqkees : thanx! that did the trick!!!
@proddy : _swap to ESP32_: will do. But first I have to practice my solder skills. It has been a while ago and at least I have to find my soldering iron first... :-)

closed 2021-09-02 by bartplessers

proddy, 2021-09-06:
@bartplessers I'm sure if you send it back to Kees he'll solder it for free :-)

bartplessers, 2021-09-11:
Hi @proddy , 

I have saved @bbqkees to solder my ESP32, and finally found some time to do it myself.
Things went well and running v3.2.1 now on ESP32 / v3.3.5-1-g85c43024c

thanx for your mental support 

grtz
B

proddy, 2021-09-12:
That's awesome news @bartplessers ! I hope you like all the new features