emsesp/EMS-ESP #775: Home Assistant MQTT Auto Discovery only discovering dallas sensor

RobbeM1991, 2021-09-05:
I've got a Gateway Premium II (P2), running esp8266 / 2.2.2-dev(38a443e).
Running Home Assistant 2021.8.8.
With addon: Mosquitto broker Current version: 6.0.1

I've followed the explanation at https://bbqkees-electronics.nl/wiki/gateway/home-assistant-configuration.html to configure MQTT auto discovery.

The MQTT integration only finds a dallas sensor but can't find my boiler or thermostat.

Any tips on how to find the boiler too?
Am I doing something wrong or has something changed since the date of the tutorial?
Did I forget something?

proddy, 2021-09-06:
I'm not sure. Version 2 for the ESP8266 was disconnected quite a while ago (I would strongly suggest swapping out the ESP8266 for an ESP32 and using v3). Anyway, use MQTTExplorer and take a peek at the topics that start with 'homeassistant'. Also look into HA at the Integrations to see what is listed. There's some notes on this at https://emsesp.github.io/docs/#/Home-Assistant

bbqkees, 2021-09-07:
Did it work before? 

If you changed settings on EMS-ESP V2 save them and then disconnect the Gateway from the power completely and connect it back.

The HA guide on the wiki was created with V2 at the time so the direction should still be correct.

RobbeM1991, 2021-09-07:
> I'm not sure. Version 2 for the ESP8266 was disconnected quite a while ago (I would strongly suggest swapping out the ESP8266 for an ESP32 and using v3). Anyway, use MQTTExplorer and take a peek at [...]
[image]

I just took a look through MQTT explorer.
The boiler info is published, but not under homeassistant... Any possible reason for that?

Also, it seems that I can't configure the temperature setpoint as settable in the ems-esp web interface, would I be able to do that with the ESP32 version or would that remain the same?

bbqkees, 2021-09-07:
Remove the Gateway from power completely and reconnect it after 20 seconds.
Likely the MQTT discovery topics will be sent for the boiler/thermostat as well.

RobbeM1991, 2021-09-07:
> 
> 
> Remove the Gateway from power completely and reconnect it after 20 seconds.
> Likely the MQTT discovery topics will be sent for the boiler/thermostat as well.

Thanks! It's working now.
Pretty stupid of me not having tried that first though...

It seems I can also set the temperature of the thermostat, which is what I wanted.
The wiki (https://bbqkees-electronics.nl/wiki/gateway/home-assistant-configuration.html?highlight=settable) said that the parameter needed to have a pencil in front of it on the web interface, to be settable. In this case it doesn't.

But it's working and that's most important!

Thanks for creating this really nice device!

proddy, 2021-09-07:
great! happy that it works

bbqkees, 2021-09-07:
The pencil was introduced in V3 so it's not in the V2 firmware.

proddy, 2021-09-07:
btw there's a demo of V3 at https://ems-esp.derbyshire.nl although it's a slightly older version but still gives you an idea of the new features and settings

closed 2021-10-04 by proddy