emsesp/EMS-ESP #776: Home Assistant - State class support

KNXBroker, 2021-09-12:
**Is your feature request related to a problem? Please describe.**
Since some home assistant releases home assistant supports state_class, which is not support (yet) by EMS-ESP.

**Describe the solution you'd like**
It would be great if automatic entity creation would support state_class definition. E.g. for temperatures `state_class: measurement`

**Describe alternatives you've considered**
Manual definition / creation of home assistant entities.

**Additional context**
I know that ESP8266 is no longer supported, but I am still using it... In general the feature would be great for both V2 and V3. Is the code definition part of both V2 and V3 the same...? I could support if someone shows me the respective lines. I assume it's only an additional parameter which has to be added to the respective sensors.

proddy, 2021-09-13:
I'm not too familiar with the state_class. What are the benefits of adding it to the sensors?

Could you try it yourself manually? Use something like MQTTExplorer and modify the topic `homeassistant/sensor/ems-esp/thermostat_hc1_currtemp/config` and add `"state_class: measurement"`. No need to change the code. If it does what you hoped we can add it, it's not hard like you said. 

Also we need to think about which other sensors need modifying. There can be more than 200+ on a system with solar, mixing and multiple heating circuits.

Lastly, see if you can upgrade to V3 :-)

KNXBroker, 2021-09-13:
> Lastly, see if you can upgrade to V3 :-)
It's the external power supply which prevents me upgrading, unfortunately my boiler does not have a service Jack :-(

> I'm not too familiar with the state_class. What are the benefits of adding it to the sensors?
`state_class: measurement` for long time storage and `state_class: total_increasing` for (gas) energy monitoring. 

I found a way to add the attributes by customizing, but the state_class is going to be elementary. Native support by ems-esp would be good. Do you have a 100% sensor list, e.g. in excel format? I could make a validation.

proddy, 2021-09-13:
All the device values have a Unit Of Measure which can be:

  NONE
  DEGREES
  PERCENT
  LMIN
  KWH
  WH
  HOURS
  MINUTES
  UA
  BAR
  KW
  W
  KB
  SECONDS
  DBM
  NUM
  BOOLEAN
  LIST
  MV

and these are used for rendering and also setting the default icon in HA. I'm guessing all them need the state_class 'measurement' apart from LIST, HOURS and MINUTES?

KNXBroker, 2021-09-13:
Unfortunately decision has to be made on sensor level, see https://developers.home-assistant.io/docs/core/entity/sensor/

Example: A sensor representing a setting (e.g. selected flow Temperatur in degree) is no measurement and may contain no sensor class.

I'm Not Sure about hours and minutes (e.g. operating hours). Could be total_increasing but I have to ask / try.

proddy, 2021-09-13:
It says "**_The state represents a measurement in present time, not a historical aggregation such as statistics or a prediction of the future_**" so in theory all the device values captured on the EMS fall into this category as they are actual live values, and not calculations. 

I can add to the code. But still not sure what the difference will be in HA or what the advantage is?

KNXBroker, 2021-09-13:
`state_class:total_increasing` enables sensor usage for the new energy dashboard, see https://www.home-assistant.io/blog/. This state class is for increasing energy counters like gas and power.

`state_class: measurement`enables long time recording. For the time being benefit is limited (e.g. creation of long term statistic graphs), but additional features shall be added.

The `state_class` is only for sensors, not binary sensors.

proddy, 2021-09-13:
Ok I've made the change in V3 and will monitor it for a few hours. if it works I'll backport it to v2.

KNXBroker, 2021-09-14:
Amazing reactivity 😃  One comment, I'm not sure if Wh and KWh are measurements. Unfortunately I have only a very simple boiler without solar, therefore I cannot test this.

If have checked the code (not sure if I missed sensors):

For KWH there are the following sensors, All sensors seems to be `state_class:total_increasing`
- boiler: nrgConsTotal
- boiler: auxElecHeatNrgConsTotal_
- solar : energyTotal:

For WH there are the following sensors, all sensors seems to be `state_class:none`
- WH / Solar: energyLastHour: 
- WH / Solar: energyToday

proddy, 2021-09-14:
agree, I'll remove the 'measurement' for Wh, KWh, W and KW

MichaelDvP, 2021-09-14:
I think it's a good start but a bit more complicated.
> A sensor representing a setting (e.g. selected flow Temperatur in degree) is no measurement and may contain no sensor class.

For that all values with `dv.has_cmd` should not have the `measurement` flag, The `has_cmd` has to be parsed to `publish_mqtt_ha_sensor(` 

`state_class:total_increasing` is in my understanding all `#starts`  and all `upTimes`, so `DeviceUOM::NUM` and `DeviceUOM::MINUTES` can be flaged this way. Fortunately the settings with minutes like `pumpDelay` have a command and should be skipped.  

The powervalues `W` and `kW` are measurements, representing an actual value. Energy `kWh` is total increasing for the total values, not for `today` and `last_hour`, but these are measured in `Wh`. I'm not sure, but i think `energy_last_hour` is a measurement, giving on each publish the value of the past 60min (not resetting on every full hour).

KNXBroker, 2021-09-14:
> A sensor representing a setting (e.g. selected flow Temperatur in degree) is no measurement and may contain no sensor class.

This was my  personal understanding, but I'm not 100% sure. Is a setting a measurement?!

Regarding `energy_last_hour`, I'm neither sure.

Maybe both a question for Home Assistant Community.

MichaelDvP, 2021-09-14:
> This was my personal understanding, but I'm not 100% sure. 

Oh, i thought it was from the HA spec.

> Is a setting a measurement?!

I don't use HA and don't know what this `state_class` is used for. It was only a hint how we can manage it to exclude the settings.

proddy, 2021-09-14:
it does look a little more tricky. I'm still not 100% sure of the value this brings to the end user in HA or what really happens under the sheets within HA.

KNXBroker, 2021-09-16:
For the time being the main advantage is long time storage. In the past this was only possible with external databases like INFLUX / GARFANA. Here an example of my outside temperature.
[image]

Regarding `energy_last_hour`, this is a `measurement` (If have asked on Home Assistant discord server). 

Regarding settings, I think that's not cleary defined, but after thinking about it once more, why not declaring it as `measurement`. In the end it's a current value and it (could) make sense to make a long time storage.

proddy, 2021-09-16:
right, most people use influx. I thought if you wanted to store more information you just change the 'recorder' and add more days.

As Michael mentioned it's not as easy as stating every type EMS-ESP is either a measurement or a total_increasing. Could you specify which values you are particularly interested in having long-term retention for and we'll start with those? For example the temperatures?

KNXBroker, 2021-09-16:
> I thought if you wanted to store more information you just change the 'recorder' and add more days.

My understanding is, that this will kill one-day home assistant recorder database. Long-term storage does some kind if compression, e.g. calculation max, min and mean values per hour.

Thank you for your offer, but there is no pressure including the feature only for me. I did add the state class manually via 'customizing'. I thought it would be a nice improvement for EMS-ESP in general, therefore I have started the discussion. I appreciate open-source products like yours, this was / is my way of participating.  If additional requests are coming, the feature can be fully implemented one day.

proddy, 2021-09-24:
moving this to EMS-ESP32 for implementation.