emsesp/EMS-ESP #777: upgrade EMS-ESP breaks my historical graphs in grafana

bartplessers, 2021-09-16:
**Bug description**

In HomeAssistant, I configured InfluxDB and Grafana to get the boiler data and show them on a graph.
I upgraded from EMS-ESP v2.2.1 to v3.2.1
No I notice that some of the measurements are missing

**Expected behavior**
The entities will have same name after upgrade

**Screenshots**
1. before upgrade
2. after upgrade

[image]

**Device information**
```
{
"System": {
"version": "3.2.1",
"uptime": "004+13:27:31.642",
"uptime_sec": 394051,
"freemem": 177
},
"Status": {
"bus": "connected",
"bus protocol": "Buderus",
"telegrams received": 1010860,
"read requests sent": 105115,
"write requests sent": 3,
"incomplete telegrams": 3,
"tx fails": 0,
"rx line quality": 100,
"tx line quality": 100,
"MQTT publishes": 59441,
"MQTT publish fails": 0,
"dallas sensors": 0,
"dallas reads": 0,
"dallas fails": 0
},
"Devices": [
{
"type": "Boiler",
"name": "Buderus Logamax Plus/GB192/Condens GC9000 (DeviceID:0x08 ProductID:208, Version:01.04)",
"handlers": "0x10 0x11 0x14 0x15 0x1C 0x18 0x19 0x1A 0x35 0x16 0x33 0x34 0x26 0x2A 0xD1 0xE3 0xE4 0xE5 0xE6 0xE9 0xEA"
},
{
"type": "Thermostat",
"name": "RC300/RC310/Moduline 3000/1010H/CW400/Sense II (DeviceID:0x10, ProductID:158, Version:74.01)",
"handlers": "0xA3 0x06 0xA2 0x12 0x2A5 0x2B9 0x2AF 0x29B 0x2A6 0x2BA 0x2B0 0x29C 0x2A7 0x2BB 0x2B1 0x29D 0x2A8 0x2BC 0x2B2 0x29E 0x2F5 0x31B 0x31D 0x31E 0x23A 0x267 0x240"
},
{
"type": "Gateway",
"name": "KM200/MB LAN 2 (DeviceID:0x48, ProductID:189, Version:04.07)",
"handlers": ""
},
{
"type": "Controller",
"name": "9000i (DeviceID:0x09, ProductID:224, Version:35.06)",
"handlers": ""
}
]
}

```

**Question:
What caused this behavior?
Can it be avoided in the future?
How to resolve it?

kind regards,
Bart

MichaelDvP, 2021-09-17:
V3 has changed some names of values and commands. Now all are lower case. In V2 there was a mixture of lower and mixed case and some commands have different names to the value. [Read here](https://github.com/emsesp/EMS-ESP32/issues/35).
The actual names in v3 you can [lookup here](https://github.com/emsesp/EMS-ESP32/blob/db34785be08cb765d821519722f5ce95578f6b46/src/locale_EN.h#L378).

proddy, 2021-09-18:
I think Bart's issue is the entity names in Home Assistant which are formed from the short names. For example `curflowtemp` is `sensor.boiler_current_flow_temperature` in HA. If we change any full names in `locale_EN.h` it will change the entities in HA and break any custom scripts/triggers people may have and annoy them.

Perhaps I can create a new command and endpoint like /api/<device type>/ha which lists the mappings, similar to how http://ems-esp/api/boiler/commands works.