emsesp/EMS-ESP #779: Boiler -> heatingActivated how to?

kempa007, 2021-09-25:
Hi,

I'm running ems-esp E32 board, and slowly everything is working, but I can't figure out how to set boiler heatingActivated variable wih my OpenHab MQTT binding.
This is the log:

`Received boiler => {cmd:"heatingactivated", data: 0, id="123"} (length 43)
002+09:18:48.481 E 6: [mqtt] MQTT error: payload cmd, error InvalidInput
`

and this is my things file:
`Type switch : boiler_heating_cmd "Wlaczenie/wylacznie ogrzewania" [commandTopic="ems-esp/boiler", formatBeforePublish="{cmd:\"heatingactivated\", data: %d, id=\"123\"}"]
`
First of all I would like to make sure that I'm using the right id. I assume, that id is the right id for my boiler?

This is my api/boiler:

<html><body>
<div class="toolbar"><div class="devtools-separator"></div><input class="searchBox devtools-filterinput" placeholder="Filtruj JSON"></div><div class="panelContent">

  |  
-- | --
id | 123
heatingactive | 0
tapwateractive | 0
selflowtemp | 27
selburnpow | 100
heatingpumpmod | 0
outdoortemp | 16
curflowtemp | 56.7
switchtemp | 22.6
burngas | 0
flamecurr | 0
heatingpump | 0
fanwork | 0
ignwork | 0
heatingactivated | 1

</div>
</body>
</html>

proddy, 2021-09-25:
The documentation I just realized for MQTT is very outdated (I'll get to updating that at some point) but the basics are still valid at https://emsesp.github.io/docs/#/MQTT?id=sending-commands

from your specific use case, `data` is either 1 or 0 (or on/off or true/false) and you don't need the `id` as it's really only used for the Thermostat when identifying multiple heating circuits.

you may also need to put quotes around the `cmd` and `data` but I'm not sure - depends on the client

which EMS-ESP version are you using?

kempa007, 2021-09-25:
Yes, you helped me few days ago with setting up thermostat temperatures. So I upgradeed to the beta you did recommend. So I can just delete id field or leave it empty?

The quotes I just copied from thermostat commands which are working so should be okay.

proddy, 2021-09-25:
I just tested  `topic="ems-esp/boiler" payload="{\"cmd\":\"heatingactivated\",\"data\":1}"` and it works?

kempa007, 2021-09-25:
Thanks, now it works too:

Type switch : boiler_heating_cmd "Wlaczenie/wylacznie ogrzewania" [commandTopic="ems-esp/boiler", formatBeforePublish="{cmd:\"heatingactivated\", data: %s}", on="1", off="0" ] (this is Thing file form OpenHab 3)

Leaving it if someone would have the same problem. Id field was making a problem- I thing so.

proddy, 2021-09-25:
the id is ignored so it doesn't matter what value you use with that boiler command. 

Your error was a typo  `id=\"123\"` instead of `id:123`  (the `=` should be a `:` )

kempa007, 2021-09-25:
Yes, i saw it too and corrected. Will check tomorrow again for sure.

kempa007, 2021-09-26:
So I checked command with id=\"\" and it is not working. It's very sensitive. 

Now I try to find way to do same way with warm water activated value and there are another issues.

I did exactly the same why as with heating, but I got the log:
`
Executing the JSONPATH-transformation failed: Invalid path '$.wwactivated' in '{"id":123,"heatingactive":1,"tapwateractive":0,"selflowtemp":32,"selburnpow":100,"heatingpumpmod":10,"outdoortemp":13.3,"curflowtemp":35,"switchtemp":25.8,"burngas":1,"flamecurr":15.5,"heatingpump":1,"fanwork":1,"ignwork":0,"heatingactivated":1,"heatingtemp":55,"pumpmodmax":100,"pumpmodmin":10,"pumpdelay":3,"burnminperiod":10,"burnminpower":0,"burnmaxpower":100,"boilhyston":-6,"boilhystoff":6,"curburnpow":25,"burnstarts":13117,"burnworkmin":278663,"heatworkmin":238351,"ubauptime":1055447,"servicecode":"-H","servicecodenumber":200}'
`

In the above JSON statement I don't even see wwActivated field. Is it okay or something is missing?

kempa007, 2021-09-26:
I was using boiler_data instead boiler_data_ww, but still cant read warm water activated state:

Type switch : boiler_ww_state "Stan wlaczenia CWU" [stateTopic="ems-esp/boiler_data_ww", transformationPattern="JSONPATH:$.wwactivated", on="1", off="0" ]

proddy, 2021-09-26:
it does work with the `id` - I added it to the test suite and the id is just ignored.

To debug, start using the WebUI to make sure the data is there and writable. Also use the console for debugging. And also use the API to get some values, like http://ems-esp.local/api/boiler/commands will list the available commands (it will show wwactivated).

I think the problems are in your OpenHAB implementation. Best if you consult BBQKees who wrote https://bbqkees-electronics.nl/wiki/gateway/openhab-configuration.html

kempa007, 2021-09-26:
Yeah, sorry, another typo. It works, I had just use boiler_data_ww topic.

kempa007, 2021-09-26:
Before you close the topic, I would like to ask you how can I test commands to be sure that they are okay.

proddy, 2021-09-26:
go to console, type `show commands`. To call one do first `su` then use `call`, e.g. `call boiler activated`. You can hit <TAB> for auto-complete. All this is documented by the way

closed 2021-10-04 by proddy